jsk-ros-pkg/jsk_3rdparty #10: sourceforce is down and cannot install libsiftfast

garaemon, 2015-07-20:
http://libsift.sourceforge.net/

long long downtime. we should move libsiftfast to github.

garaemon, 2015-07-20:
see http://sourceforge.net/blog/sourceforge-infrastructure-and-service-restoration/

k-okada, 2015-07-20:
This is exactly I discussed in
https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/8, and fundamental problem
is keeping sourcecode by our selves (in this case, we keep them as
downloading from package) , so our option is
- patience, wait until source forge is back
- reliease libsift directory from -release repo

If you wan to merge https://github.com/jsk-ros-pkg/jsk_common/pull/1035,
feel free to merge, we all know why travis did not passed on deb setup

garaemon, 2015-07-20:
About libsiftfast, I think the best way is to move it from sourceforge to jsk-ros-pkg/libsiftfast (sourceforge is less reliable than github) and release it as non-catkin package (in order not to manage building by ourselves).

k-okada, 2015-07-20:
but we're not owner of libsiftfast

garaemon, 2015-08-06:
> but we're not owner of libsiftfast

hmm, it's a LGPL software.

garaemon, 2015-08-06:
but sourceforge is back now, I will close the issue. If we have another problem about sourceforge, it would be a time to think about migration.

closed 2015-08-06 by garaemon