jsk-ros-pkg/jsk_3rdparty #11: laser_filters_jsk_patch is not used

garaemon, 2015-08-06:
laser_filters_jsk_patch is not used anymore. Can we remove it?

```
$ grep -iIrn "ScanFootObjectFilter" *                                               
grep: jsk-ros-pkg/jsk_recognition/jsk_pcl_ros/src/tool/.#icp_registration_viewer.cpp: そのようなファイルやディレクトリはありません
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/launch/footobject_filter_example.yaml:3:  type: laser_filters/ScanFootObjectFilter
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/src/laser_scan_filters.cpp:37:PLUGINLIB_DECLARE_CLASS(laser_filters, ScanFootObjectFilter, laser_filters::ScanFootObjectFilter, filters::FilterBase<sensor_msgs::LaserScan>)
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/laser_filters_plugins.xml:3:    <class name="laser_filters/ScanFootObjectFilter" type="laser_filters::ScanFootObjectFilter"
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/include/laser_filters_jsk_patch/scan_footobject_filter.h:48:/** @b ScanFootObjectFilter is a simple filter that filters shadow points in a laser scan line 
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/include/laser_filters_jsk_patch/scan_footobject_filter.h:51:class ScanFootObjectFilter : public filters::FilterBase<sensor_msgs::LaserScan>
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/include/laser_filters_jsk_patch/scan_footobject_filter.h:60:  ScanFootObjectFilter () 
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/include/laser_filters_jsk_patch/scan_footobject_filter.h:86:  virtual ~ScanFootObjectFilter () { }
jsk-ros-pkg/jsk_3rdparty/jsk_ros_patch/laser_filters_jsk_patch/include/laser_filters_jsk_patch/scan_footobject_filter.h:145:    ROS_DEBUG("ScanFootObjectFilter  %d Points from scan with min radius: %.2f, num of pints: %d", (int)indices_to_publish.size(), max_radius_, min_points_);
grep: rtm-ros-robotics/rtmros_hrp2/jsk_hrp2_ros_bridge/patch: そのようなファイルやディレクトリはありません
grep: rtm-ros-robotics/rtmros_hrp2/hrp2_models/HIRO/.#main.wrl: そのようなファイルやディレクトリはありません
```

k-okada, 2015-08-06:
https://github.com/jsk-ros-pkg/jsk_demos/blob/03ced369035e29bbcad3451eae3ea017fb084c68/jsk_2013_04_pr2_610/launch/detect_footobject.launch

did you sent PR to upstream @aginika , if not please do that ASAP

aginika, 2015-08-10:
@k-okada About inf problem around laser_fileters?
If so, I sent PR a little while ago.

k-okada, 2015-08-10:
Do they merged your PR?

aginika, 2015-08-10:
Not yet, cause I sent the PR 2 hours ago.

k-okada, 2015-08-10:
could you write the link to your PR in this thread?

aginika, 2015-08-10:
https://github.com/ros-perception/laser_filters/pull/27

k-okada, 2015-08-10:
Btw, when we write https://github.com/jsk-ros-pkg/jsk_demos/blob/03ced369035e29bbcad3451eae3ea017fb084c68/jsk_2013_04_pr2_610/launch/detect_footobject.launch, we did not use gazebo. So I thin the reason we need this is different than you described in https://github.com/ros-perception/laser_filters/pull/27
Or do we have same nan/inf problem on real robot?

aginika, 2015-10-28:
no, nan/inf problem is only in Gazebo.

jonbinney, 2022-03-21:
I noticed this laser_filters_jsk_patch package on the buildfarm and was wondering why it needed to exist. I'm maintaining the laser_filters package; let me know if I can help fix any problems with the laser_filters_package.  Looks like the upsream PR got merged.  Would be nice to get rid of the patch package, it would save some build time on the farm.

k-okada, 2022-03-22:
@jonbinney thank you, all of our patches are merged in upstream, so we do not need to maintain laser_filters_jsk_patch, so removed from release repository, see https://github.com/tork-a/jsk_3rdparty-release/pull/1

closed 2022-03-22 by k-okada