jsk-ros-pkg/jsk_3rdparty #14: fails tweeting with image on PR2

furushchev, 2015-08-20:
pr2 often fails to tweet with image:
https://twitter.com/pr2jsk/status/634221745029820416

```
Amazing! Furushchev has used PR2 for 55 minutes, I'm not tired!
(image not published)
```

`rostwitter` uses `/tmp` to store image published, but PR2 does not share `/tmp` on 2 PCs (c1 and c2), because `rostwitter` could not find image tweeted on another pc.

may be good to store image on `$HOME/.rostwitter`?

garaemon, 2015-08-20:
Who does save the image?

furushchev, 2015-08-20:
Or scp image to host which executes tweet.py when needs?
(maybe need to change `jsk_robot_start/lifelog/tweet_client.l`)

k-okada, 2015-08-20:
so you need to run iamge_saver at the machine runnning rostwitter

furushchev, 2015-08-20:
@garaemon every nodes which publishes `/tweet` topic.

furushchev, 2015-08-20:
@k-okada ok. I'll create PR.

k-okada, 2015-08-20:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pepper_robot/jsk_pepper_startup/launch/jsk_pepper_startup.launch#L34

k-okada, 2015-08-20:
I don think this is rostwitter problem, but pr2 launch file is not correct. see papper case, running rostwitter and image_save at same machine and https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pepper_robot/jsk_pepper_startup/nodes/tweet.l just calls to save images 

(the papper code is same as https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_robot_common/jsk_robot_startup/lifelog/tweet.l#L18, so we need to merge them)

garaemon, 2015-08-20:
How about saving image inside of rostwitter instead of using image_saver?

furushchev, 2015-08-20:
is there no problem if we remove the feature to read image topic from parameter `/active_user/image_topic`?

k-okada, 2015-08-20:
On Thu, Aug 20, 2015 at 4:20 PM, Furushchev <notifications@github.com> 
wrote:

> the feature to read image topic from parameter /active_user/image_topic?
>

what it this? who is reading this topic?

--
◉ Kei Okada

furushchev, 2015-08-20:
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_robot_common/jsk_robot_startup/lifelog/tweet_client.l#L5-L8

k-okada, 2015-08-20:
oh, you can change tweet_client.l to use image_saver, not extract_images

closed 2015-09-10 by furushchev