jsk-ros-pkg/jsk_3rdparty #21: Failed to build bayesian_belief_network

garaemon, 2015-09-08:
http://jenkins.ros.org/job/doc-jade-jsk_3rdparty/18/

```
running build
running build_py
error: package directory 'lib/python2.7/dist-packages/bayesian' does not exist
CMake Error at jsk_3rdparty/bayesian_belief_networks/catkin_generated/safe_execute_install.cmake:4 (message):

  execute_process(<http://jenkins.ros.org/job/doc-jade-jsk_3rdparty/ws/doc_stacks/2015-09-08_13-29-58.198989/build_repo/jsk_3rdparty/bayesian_belief_networks/catkin_generated/python_distutils_install.sh)>
  returned error code
Call Stack (most recent call first):
  jsk_3rdparty/bayesian_belief_networks/cmake_install.cmake:36 (INCLUDE)
  cmake_install.cmake:130 (INCLUDE)

make: *** [install] Error 1
/!\  Failed to execute command '['make', 'install']' with return code 2
FAILED TO CALL MAKE INSTALL ON CATKIN REPOS
Exception:
I: Copying back the cached apt archive contents
I: unmounting /var/cache/pbuilder/ccache filesystem
I: unmounting <http://jenkins.ros.org/job/doc-jade-jsk_3rdparty/ws/> filesystem
I: unmounting /home/rosbuild filesystem
I: unmounting dev/pts filesystem
I: unmounting proc filesystem
I: cleaning the build env
I: removing directory /var/cache/pbuilder/build//14924 and its subdirectories
Build step 'Execute shell' marked build as failure
Build step 'Groovy Postbuild' marked build as failure
Recording test results
```

garaemon, 2015-09-08:
Is it only about document generation? can we ignore this error?

k-okada, 2015-09-08:
Yes, you can ignore now

closed 2015-09-08 by garaemon