jsk-ros-pkg/jsk_3rdparty #39: [nlopt] build error on indigo

wkentaro, 2015-11-17:
```
[nlopt] ./nlopt_optimize-oct.cc:32:31: error: Octave_map was not declared in this scope                                        
[nlopt]  static int struct_val_default(Octave_map &m, const std::string& k,                                                    
[nlopt]                                ^                                                                                       
[nlopt] ./nlopt_optimize-oct.cc:32:43: error: m was not declared in this scope                                                 
[nlopt]  static int struct_val_default(Octave_map &m, const std::string& k,                                                    
[nlopt]                                            ^                                                                           
[nlopt] ./nlopt_optimize-oct.cc:32:46: error: expected primary-expression before const                                         
[nlopt]  static int struct_val_default(Octave_map &m, const std::string& k,                                                    
[nlopt]                                               ^                                                                        
[nlopt] ./nlopt_optimize-oct.cc:33:6: error: expected primary-expression before int                                            
[nlopt]       int dflt)                                                                                                        
[nlopt]       ^                                                                                                                
[nlopt] ./nlopt_optimize-oct.cc:33:14: error: expression list treated as compound expression in initializer [-fpermissive]     
[nlopt]       int dflt)                                                                                                        
[nlopt]               ^                                                                                                        
[nlopt] ./nlopt_optimize-oct.cc:34:1: error: expected , or ; before { token                                                    
[nlopt]  {                                                                                                                     
[nlopt]  ^                                                                                                                     
[nlopt] make[7]: *** [nlopt_optimize.oct] Error 1                                                                              
[nlopt] make[6]: *** [all] Error 2                                                                                             
[nlopt] make[5]: *** [all-recursive] Error 1                                                                                   
[nlopt] make[4]: *** [all] Error 2                                                                                             
[nlopt] make[3]: *** [installed] Error 2                                                                                       
[nlopt] make[2]: *** [/home/wkentaro/ros/indigo/devel/lib/libnlopt_cxx.so] Error 2                                             
[nlopt] make[1]: *** [CMakeFiles/libnlopt_cxx.dir/all] Error 2                                                                 
[nlopt] make: *** [all] Error 2                                                                                                
[nlopt] <== '/home/wkentaro/ros/indigo/build/nlopt/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2' 
Failed   <== nlopt [ 3.9 seconds ]                                                                                             
[build] There were '1' errors:                                                                                                 

Failed to build package 'nlopt' because the following command:

```

k-okada, 2015-11-18:
assigned to @s-noda

k-okada, 2015-11-18:
it passed on travis, so wich compiler are you using?
https://travis-ci.org/jsk-ros-pkg/jsk_3rdparty/jobs/91743953

wkentaro, 2015-11-18:
gcc? https://gist.github.com/wkentaro/bab6aaa8124462f9804a#file-nlopt_build_log-txt-L58

s-noda, 2015-11-18:
> [nlopt]  static int struct_val_default(Octave_map &m, const std::string& k,                                                    

I think this line means that octave is missing.
So, could you find octave directory under the 3rdparty/nlopt/build/nlopt-x.x ?
If not, remove build dir, and try to build again?

wkentaro, 2015-11-18:
need to extract nlopt-x.x.tgz manually?

https://gist.github.com/wkentaro/475816be9008f395ccf2

wkentaro, 2015-11-18:
I tried but had error.
https://gist.github.com/d092db66e96471e8b4be

s-noda, 2015-11-18:
> need to extract nlopt-x.x.tgz manually?

I believe no.

In your environment, octave is installed?
nlopt seems to depend on octave, but there might be no installation of octave.

I think `sudo aptitude install octave` is ok.

wkentaro, 2015-11-18:
octave is installed
```
% dpkg --get-selections | grep octave
liboctave-dev					install
liboctave2:amd64				deinstall
liboctave3:amd64				install
octave						install
octave-common					install
```

I also reinstalled these.

wkentaro, 2015-11-18:
this happens on my laptop and desktop pc.
```
# desktop
% dpkg --get-selectiosn | grep octave
liboctave-dev					install
liboctave3:amd64				install
octave						install
octave-common					install
```

s-noda, 2015-11-18:
I have only hydro environment.
In my environments, this  octave dir is dummy compiled.

I'm not confident, but, 
this octave compile is switched by using some system environment variables such as WITH_OCTAVE_TRUE, OCT_INSTALL_DIR,,, in configure script of nlopt.

In my environment, `env | grep OCT` is empty.

wkentaro, 2015-11-18:
I think why travis test passes is there is no octave compilation on that. (It can detect octave installation)
TRAVIS: 
```
[nlopt] checking for Python.h... yes
[nlopt] checking for Numpy include directory... /usr/lib/python2.7/dist-packages/numpy/core/include
[nlopt] checking for numpy/arrayobject.h... yes
[nlopt] checking for mkoctfile... no
[nlopt] configure: WARNING: can't find mkoctfile: won't be able to compile GNU Octave plugin
[nlopt] checking for mex... no
[nlopt] configure: WARNING: can't find mex: won't be able to compile Matlab plugin
[nlopt] checking for working HUGE_VAL... ok
[nlopt] checking that generated files are newer than configure... done
[nlopt] configure: creating ./config.status
[nlopt] config.status: creating Makefile
[
```
ON MY PC: https://gist.github.com/wkentaro/bab6aaa8124462f9804a#file-nlopt_build_log-txt-L183-L185

wkentaro, 2015-11-18:
> In my environment, env | grep OCT is empty.

Mine also.

wkentaro, 2015-11-18:
> I think why travis test passes is that there is no octave compilation on it. (It can detect octave installation)

the error disappears after removing octave from environment.

```
sudo aptitude purge octave
```

wkentaro, 2015-11-18:
If no one is using nlopt on indigo.
We can skip nlopt at build time with some changes on CMakeLists.txt.

s-noda, 2015-11-18:
great!

`We can skip nlopt at build time with some changes on CMakeLists.txt.`

I think it's ok, but, please check the following diff?
I think this option `--without-octave` skip the octave compile.

```
diff --git a/3rdparty/nlopt/Makefile b/3rdparty/nlopt/Makefile
index 1db4cd2..fb7de39 100644
--- a/3rdparty/nlopt/Makefile
+++ b/3rdparty/nlopt/Makefile
@@ -10,7 +10,7 @@ MK_DIR = $(shell rospack find mk)
 include $(MK_DIR)/download_unpack_build.mk
 
 installed: $(SOURCE_DIR)/unpacked
-       (cd $(SOURCE_DIR) && ./configure --enable-shared --with-cxx --prefix=$(DSTDIR) LIBS='-lstdc++' && make && make install)
+       (cd $(SOURCE_DIR) && ./configure --enable-shared --with-cxx --prefix=$(DSTDIR) LIBS='-lstdc++' --without-octave && make && make install)
        touch installed
 clean:
        -rm -rf include lib share $(SOURCE_DIR) installed
```

wkentaro, 2015-11-18:
That fixed the problem.

s-noda, 2015-11-18:
Thanks.

I believe this update has no effect on my programs
because this octave option seems to compile the octave wrapper of nlopt 
and this wrapper has no user in JSK.

k-okada, 2015-11-18:
thanks!!!! @wkentaro @s-noda

PR #41 (2015-11-18): [nlopt] Stop compiling with octave
Fixes #39
This is because there is an error while compiling nlopt with octave on
Indigo Ubuntu 14.04.
commit 1a9b1f08 (2015-11-18): [nlopt] Stop compiling with octave

Fixes #39
This is because there is an error while compiling nlopt with octave on
Indigo Ubuntu 14.04.
closed 2015-11-18 by k-okada