jsk-ros-pkg/jsk_3rdparty #42: cannot compile libsiftfast

garaemon, 2015-11-23:
```
$ catkin b libsiftfast -iv
==> Expanding alias 'b' from 'catkin b libsiftfast -iv' to 'catkin build libsiftfast -iv'
--------------------------------------------------------------------------------
Profile:                     default
Extending:          [cached] /home/garaemon/ros/hydro_parent/devel:/opt/ros/hydro
Workspace:                   /home/garaemon/ros/hydro
Source Space:       [exists] /home/garaemon/ros/hydro/src
Build Space:        [exists] /home/garaemon/ros/hydro/build
Devel Space:        [exists] /home/garaemon/ros/hydro/devel
Install Space:     [missing] /home/garaemon/ros/hydro/install
DESTDIR:                     None
--------------------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
--------------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
--------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------------------------- 
Found '298' packages in 0.5 seconds. 
Starting ==> libsiftfast                                                        
[libsiftfast] ==> '/home/garaemon/ros/hydro/build/libsiftfast/build_env.sh /usr/bin/make cmake_check_build_system' in '/home/garaemon/ros/hydro/build/libsiftfast' 
[libsiftfast] <== '/home/garaemon/ros/hydro/build/libsiftfast/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0' 
[libsiftfast] ==> '/home/garaemon/ros/hydro/build/libsiftfast/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/garaemon/ros/hydro/build/libsiftfast' 
[libsiftfast] [100%] Generating /home/garaemon/ros/hydro/devel/lib/libsiftfast.so 
[libsiftfast] make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule. 
[libsiftfast] At revision 54.                                                   
[libsiftfast] Reverted 'CMakeLists.txt'                                         
[libsiftfast] patching file CMakeLists.txt                                      
[libsiftfast] -- Using cmake version 2.8                                        
[libsiftfast] -- Using SSE2 extensions                                          
[libsiftfast] -- found boost version: 104601                                    
[libsiftfast] -- installing to /home/garaemon/ros/hydro/devel                   
[libsiftfast] -- python and boost-python found                                  
[libsiftfast] -- compiling with openmp support                                  
[libsiftfast] -- MATLAB installation not found                                  
[libsiftfast] -- Octave installation not found                                  
[libsiftfast] -- Configuring done                                               
[libsiftfast] -- Generating done                                                
[libsiftfast] -- Build files have been written to: /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/build 
[libsiftfast] [ 16%] Built target libsiftfast                                   
[libsiftfast] [ 33%] Built target libsiftfast-static                            
[libsiftfast] [ 66%] Built target siftfast                                      
[libsiftfast] [ 83%] Building CXX object CMakeFiles/siftfastpy.dir/siftfastpy.o 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function static void* int_from_int::convertible(PyObject*): 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:355:44: error: PyNumber_Int was not declared in this scope 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:356:32: error: PyString_Check was not declared in this scope 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function static void int_from_int::construct(PyObject*, boost::python::converter::rvalue_from_python_stage1_data*): 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:371:45: error: PyNumber_Int was not declared in this scope 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function static void* T_from_number<T>::convertible(PyObject*): 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:390:32: error: there are no arguments to PyString_Check that depend on a template parameter, so a declaration of PyString_Check must be available [-fpermissive] 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:390:32: note: (if you use -fpermissive, G++ will accept your code, but allowing the use of an undeclared name is deprecated) 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In function void init_module_siftfastpy(): 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:415:5: error: return-statement with a value, in function returning 'void' [-fpermissive] 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function static void* T_from_number<T>::convertible(PyObject*) [with T = float, PyObject = _object]: 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:384:9:   instantiated from T_from_number<T>::T_from_number() [with T = float] 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:419:26:   instantiated from here 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:390:9: error: PyString_Check was not declared in this scope 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function static void* T_from_number<T>::convertible(PyObject*) [with T = double, PyObject = _object]: 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:384:9:   instantiated from T_from_number<T>::T_from_number() [with T = double] 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:420:27:   instantiated from here 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:390:9: error: PyString_Check was not declared in this scope 
[... 4 lines cut ...]
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function static void* T_from_number<T>::convertible(PyObject*) [with T = double, PyObject = _object]: 
[libsiftfast] /home/garaemon/ros/hydro/build/libsiftfast/libsiftfast_svn/siftfastpy.cpp:401:5: warning: control reaches end of non-void function [-Wreturn-type] 
[libsiftfast] make[6]: *** [CMakeFiles/siftfastpy.dir/siftfastpy.o] Error 1     
[libsiftfast] make[5]: *** [CMakeFiles/siftfastpy.dir/all] Error 2              
[libsiftfast] make[4]: *** [all] Error 2                                        
[libsiftfast] make[3]: *** [libsiftfast] Error 2                                
[libsiftfast] make[2]: *** [/home/garaemon/ros/hydro/devel/lib/libsiftfast.so] Error 2 
[libsiftfast] make[1]: *** [CMakeFiles/libsiftfast.dir/all] Error 2             
[libsiftfast] make: *** [all] Error 2                                           
[libsiftfast] <== '/home/garaemon/ros/hydro/build/libsiftfast/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2' 
Failed   <== libsiftfast [ 4.4 seconds ]                                        
[build] There were '1' errors:                                                  

Failed to build package 'libsiftfast' because the following command:

# Command to reproduce:
cd /home/garaemon/ros/hydro/build/libsiftfast && /home/garaemon/ros/hydro/build/libsiftfast/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/garaemon/ros/hydro/build/build_logs/libsiftfast.log

Exited with return code: 2 
                                                                                
[build] Runtime: 5.9 seconds 
```

garaemon, 2015-11-23:
```
[libsiftfast] -- Found PythonLibs: /usr/lib/libpython3.2mu.so (found version "3.2.3") 
[libsiftfast] -- Found PythonInterp: /usr/bin/python (found version "2.7.3")    
```

garaemon, 2015-11-23:
python2.7 and 3.2 are installed
```
$ ls /usr/lib/libpython*   
/usr/lib/libpython2.7.a@      /usr/lib/libpython3.2mu.a@
/usr/lib/libpython2.7.so@     /usr/lib/libpython3.2mu.so@
/usr/lib/libpython2.7.so.1@   /usr/lib/libpython3.2mu.so.1@
/usr/lib/libpython2.7.so.1.0  /usr/lib/libpython3.2mu.so.1.0

```

wkentaro, 2015-11-23:
Please run `make` after `roscd libsiftfast` and paste the log to this issue.

wkentaro, 2015-11-23:
In my environment I cannot reproduce this, python-numpy is the main dependency to compile siftfastpy.

garaemon, 2015-11-23:
```
svn co  http://svn.code.sf.net/p/libsift/code/trunk libsiftfast_svn
A    libsiftfast_svn/LICENSE
A    libsiftfast_svn/AUTHORS
A    libsiftfast_svn/makerelease.sh
A    libsiftfast_svn/libsiftfast.cpp
A    libsiftfast_svn/README
A    libsiftfast_svn/cmake_uninstall.cmake.in
A    libsiftfast_svn/siftmex.cpp
A    libsiftfast_svn/FindOctave.cmake
A    libsiftfast_svn/profiler.cpp
A    libsiftfast_svn/siftfast.cpp
A    libsiftfast_svn/FindBoost.cmake
A    libsiftfast_svn/INSTALL
A    libsiftfast_svn/profiler.h
A    libsiftfast_svn/runcmake.bat
A    libsiftfast_svn/CMakeLists.txt
A    libsiftfast_svn/Makefile
A    libsiftfast_svn/siftfastpy.cpp
A    libsiftfast_svn/examples
A    libsiftfast_svn/examples/test1.jpg
A    libsiftfast_svn/examples/test2.jpg
A    libsiftfast_svn/examples/testsiftfast.py
A    libsiftfast_svn/examples/test0.jpg
A    libsiftfast_svn/siftfast.h.in
A    libsiftfast_svn/siftfast.m
Checked out revision 54.
if test -z "" -o "x" != "x-r `svn info libsiftfast_svn | grep Revision | cut -d " " -f 2,2`"; then \
          cd libsiftfast_svn && svn up ; \
        fi
At revision 54.
touch rospack_nosubdirs
touch patched
cd libsiftfast_svn && svn up && mkdir -p build
At revision 54.
cd libsiftfast_svn && svn revert --recursive . && patch -p0 -f -E < `rospack find libsiftfast`/patches/01.fix_python_binding.patch
patching file CMakeLists.txt
cd libsiftfast_svn/build && BOOST_INCLUDEDIR= BOOST_LIBRARYDIR= cmake -DCMAKE_INSTALL_PREFIX=`rospack find libsiftfast` -DCMAKE_BUILD_TYPE=RelWithDebInfo .. && make  install
-- The C compiler identification is GNU 4.6.4
-- The CXX compiler identification is GNU 4.6.4
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using cmake version 2.8
-- Performing Test IS_X86_64
-- Performing Test IS_X86_64 - Success
-- Performing Test HAS_SSE2_EXTENSIONS
-- Performing Test HAS_SSE2_EXTENSIONS - Success
-- Performing Test HAS_SSE_EXTENSIONS
-- Performing Test HAS_SSE_EXTENSIONS - Success
-- Using SSE2 extensions
-- Boost version: 1.46.1
-- Found the following Boost libraries:
--   system
--   python
-- found boost version: 104601
-- installing to /home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast
-- Found PythonLibs: /usr/lib/libpython3.2mu.so (found version "3.2.3")
-- Found PythonInterp: /usr/bin/python (found version "2.7.3")
-- Performing Test HAVE_ALL_PYTHON_HEADERS
-- Performing Test HAVE_ALL_PYTHON_HEADERS - Success
-- python and boost-python found
-- Performing Test HAVE_OPENMP
-- Performing Test HAVE_OPENMP - Success
-- Looking for omp_get_num_threads in gomp
[... 53 lines cut ...]
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:390:32: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In function ‘void init_module_siftfastpy()’:
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:415:5: error: return-statement with a value, in function returning 'void' [-fpermissive]
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function ‘static void* T_from_number<T>::convertible(PyObject*) [with T = float, PyObject = _object]’:
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:384:9:   instantiated from ‘T_from_number<T>::T_from_number() [with T = float]’
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:419:26:   instantiated from here
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:390:9: error: ‘PyString_Check’ was not declared in this scope
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function ‘static void* T_from_number<T>::convertible(PyObject*) [with T = double, PyObject = _object]’:
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:384:9:   instantiated from ‘T_from_number<T>::T_from_number() [with T = double]’
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:420:27:   instantiated from here
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:390:9: error: ‘PyString_Check’ was not declared in this scope
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function ‘static void* int_from_int::convertible(PyObject*)’:
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp:367:5: warning: control reaches end of non-void function [-Wreturn-type]
/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/siftfastpy.cpp: In static member function ‘static void* T_from_number<T>::convertible(PyObject*) [with T = float, PyObje [... cut ...]
make[3]: *** [CMakeFiles/siftfastpy.dir/siftfastpy.o] Error 1
make[3]: Leaving directory `/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/build'
make[2]: *** [CMakeFiles/siftfastpy.dir/all] Error 2
make[2]: Leaving directory `/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/build'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/garaemon/ros/hydro/src/jsk-ros-pkg/jsk_3rdparty/3rdparty/libsiftfast/libsiftfast_svn/build'
make: *** [libsiftfast] Error 2
```

wkentaro, 2015-11-23:
On @Tnoriaki 's PC:
```
~ ❯ ls /usr/lib/libpython*
/usr/lib/libpython2.7.a     /usr/lib/libpython2.7.so.1.0
/usr/lib/libpython2.7.so    /usr/lib/libpython3.2mu.so.1
/usr/lib/libpython2.7.so.1  /usr/lib/libpython3.2mu.so.1.0
```
no `/usr/lib/libpython3.2mu.so`

k-okada, 2015-11-23:
which compiler are you using?

garaemon, 2015-11-23:
```
$ gcc --version
gcc (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
$ g++ --version
g++ (Ubuntu/Linaro 4.6.3-1ubuntu5) 4.6.3
Copyright (C) 2011 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

garaemon, 2015-11-23:
```
$ dpkg-query -S /usr/lib/libpython3.2mu.so.1
libpython3.2: /usr/lib/libpython3.2mu.so.1
```

need to uninstall libpython3.2?
libsift's python detection does not work on the environment including python3 and python2?

```
$ sudo aptitude remove libpython3.2
The following packages will be REMOVED:  
  libpython3.2 
0 packages upgraded, 0 newly installed, 1 to remove and 12 not upgraded.
Need to get 0 B of archives. After unpacking 3,445 kB will be freed.
The following packages have unmet dependencies:
 libpyside-py3-1.1 : Depends: libpython3.2 (>= 3.2~a4) but it is not going to be installed.
 libshiboken-py3-1.1 : Depends: libpython3.2 (>= 3.2~a4) but it is not going to be installed.
The following actions will resolve these dependencies:

      Remove the following packages:         
1)      libpyside-dev                        
2)      libpyside-py3-1.1                    
3)      libshiboken-dev                      
4)      libshiboken-py3-1.1                  
5)      ros-hydro-joint-trajectory-controller
6)      ros-hydro-python-qt-binding          
7)      ros-hydro-qt-dotgraph                
8)      ros-hydro-qt-gui                     
9)      ros-hydro-qt-gui-cpp                 
10)     ros-hydro-qt-gui-py-common           
11)     ros-hydro-ros-controllers            
12)     ros-hydro-rqt-bag                    
13)     ros-hydro-rqt-console                
14)     ros-hydro-rqt-graph                  
15)     ros-hydro-rqt-gui                    
16)     ros-hydro-rqt-gui-cpp                
17)     ros-hydro-rqt-gui-py                 
18)     ros-hydro-rqt-image-view             
19)     ros-hydro-rqt-logger-level           
20)     ros-hydro-rqt-nav-view               
21)     ros-hydro-rqt-plot                   
22)     ros-hydro-rqt-py-common              
23)     ros-hydro-rqt-reconfigure            
24)     ros-hydro-rqt-robot-dashboard        
25)     ros-hydro-rqt-robot-monitor          
```

closed 2017-06-28 by k-okada