jsk-ros-pkg/jsk_3rdparty #49: Failed to release jsk_3rdparty from jsk jenkins

garaemon, 2015-12-15:
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/bloom_release/5880/console

```
17:22:24 Executing release track 'indigo'[0m
17:22:24 [34m[1m==> [0m[1mbloom-export-upstream /tmp/tmpQ4O0Or/upstream git --tag 2.0.13 --display-uri https://github.com/jsk-ros-pkg/jsk_3rdparty.git --name upstream --output-dir /tmp/tmpq1iqAU[0m[0m
17:22:24 
17:22:24 [34m[1m==> [0m[1mgit-bloom-import-upstream /tmp/tmpq1iqAU/upstream-2.0.13.tar.gz  --release-version 2.0.13 --replace[0m[0m
17:22:24 
17:22:24 [34m[1m==> [0m[1mgit-bloom-generate -y rosrelease indigo --source upstream -i 0[0m[0m
17:22:24 
17:22:24 [34m[1m==> [0m[1mgit-bloom-generate -y rosdebian --prefix release/indigo indigo -i 0[0m[0m
17:22:24 [31m[1m[31m[1m<== [0mError running command '[1m['/usr/bin/git-bloom-generate', '-y', 'rosdebian', '--prefix', 'release/indigo', 'indigo', '-i', '0']'[0m[0m[0m
17:22:24 ROS Distro index file associate with commit '7100915ee9d32105f35a1e88d381952ef6ed5693'[0m
17:22:24 New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/7100915ee9d32105f35a1e88d381952ef6ed5693/index.yaml'[0m
17:22:24 [32m[1m==> [0m[0mFetching 'jsk_3rdparty' repository from 'https://github.com/tork-a/jsk_3rdparty-release.git'[0m
17:22:24 [32m[1m==> [0mTesting for push permission on release repository[0m[0m
17:22:24 [34m[1m==> [0m[1m[0mgit remote -v[0m
17:22:24 [34m[1m==> [0m[1m[0mgit push --dry-run[0m
17:22:24 [32m[1m==> [0m[0mReleasing 'jsk_3rdparty' using release track 'indigo'[0m
17:22:24 [34m[1m==> [0m[1mgit-bloom-release indigo[0m[0m
17:22:24 [31m[1mRelease failed, exiting.[0m
17:22:24 Build step 'Execute shell' marked build as failure
17:22:24 Sending e-mails to: kei.okada+jenkins@gmail.com
17:22:25 Finished: FAILURE
```

garaemon, 2015-12-15:
Now bloom_release task does not work completely?

k-okada, 2015-12-15:
```
00:02:33.029 Applying: remove pip depends
00:02:33.029 error: patch failed: package.xml:11
00:02:33.029 error: package.xml: patch does not apply
00:02:33.029 Patch failed at 0001 remove pip depends
00:02:33.029 When you have resolved this problem run "git am --resolved".
00:02:33.029 If you would prefer to skip this patch, instead run "git am
--skip".
00:02:33.029 To restore the original branch and stop patching run "git am
--abort".
00:02:33.029
00:02:33.029 [33m [git-bloom-patch import]: Failed to apply one or more
patches for the 'Command 'git am /tmp/tmpG9x0lQ/*.patch' returned non-zero
exit status 1' branch. [0m
00:02:33.029
00:02:33.029
00:02:33.029 >>> Resolve any conflicts and when you have resolved this
problem run 'git am --resolved' and then exit the shell using 'exit 0'. <<<
00:02:33.029     To abort use 'exit 1'
00:02:33.029  [git-bloom-patch import]: User reports that conflicts have
been resolved, continuing. [0m
00:02:33.029  [git-bloom-patch import]: Applied 1 patches [0m
```
this is the problem, I'm working on this.

garaemon, 2015-12-15:
I changed ff's package.xml. It may be an issue.

k-okada, 2015-12-15:
basically bloom release does not work well on pip based packages,

garaemon, 2015-12-15:
> basically bloom release does not work well on pip based packages,

yeah, we have a lot of troubles around pip.

And also git-bloom-patch command does not work with PR-based development.
The command is kind of magic for me.

closed 2015-12-17 by garaemon