jsk-ros-pkg/jsk_3rdparty #57: .cfg files need to be fixed and released

dirk-thomas, 2016-03-20:
The .cfg files in this repo need to be fixed the same as jsk-ros-pkg/jsk_common@b79ec0a1f8638a87aea774629e465cc38ef7bd94. E.g. https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/3rdparty/mini_maxwell/cfg/RosClient.cfg

These changes need to be released. Currently the Jade jobs fail since the old cold falls back to the non-catkin code generation which won't generate the header in the correct location (e.g. http://build.ros.org/job/Jbin_uT64__mini_maxwell__ubuntu_trusty_amd64__binary/2/).

This will likely also be the same for the upcoming Indigo rebuild.

k-okada, 2016-03-20:
thaks, fixed in https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/58

dirk-thomas, 2016-03-21:
Thank you for the fast fix and release.

closed 2016-03-22 by dirk-thomas