jsk-ros-pkg/jsk_3rdparty #93: libsiftfast が1回目にビルドエラー

mukuact, 2017-04-25:
`catkin b`を始めてやると，エラーが出て，もう一度`catkin b`をすると無事通る．

makeの順番とかでしょうか？

さらの環境でのcatkin b libsiftfast -v の結果
```
mkdir catkin_test
cd catkin_test/
mkdir src
cd src
wstool init
wstool set jsk_3rdparty git@github.com:jsk-ros-pkg/jsk_3rdparty.git --git
wstool update jsk_3rdparty
cd ../
rosdep update
rosdep install --from-paths src --ignore-src -r -n -y
catkin b -v libsiftfast
```
[log.txt](https://github.com/jsk-ros-pkg/jsk_3rdparty/files/953978/log.txt)

wkentaro, 2017-04-28:
catkin-tools==0.4.X(merve-devel, link-devel)と0.3.Xの両方試しましたが、
僕の環境では起こりませんね。
あと、Travisも通っているので、環境が特殊になっているのかと思います。
(catkin-toolsがaptのじゃなかったりするかもしれません。僕の試した環境では0.4.4でした。)

```
catkin --version  # outputs 0.4.4
```

[image]

とりあえず治したければmerge-develを使うか、
```
# 0.4.X
catkin config --merge-devel
```
catkin-tools==0.3.Xを使えばいいかと思います。
```
sudo pip install 'catkin-tools==0.3.1'
```

k-okada, 2017-04-28:
普通にcatkin 入れているけど、動いているともいます．環境は以下．

```
$ catkin --version
catkin_tools 0.4.4 (C) 2014-2017 Open Source Robotics Foundation
catkin_tools is released under the Apache License, Version 2.0 (http://www.apache.org/licenses/LICENSE-2.0)
---
Using Python 2.7.6 (default, Oct 26 2016, 20:30:19) [GCC 4.8.4]
$ aptitude show python-catkin-tools
Package: python-catkin-tools             
State: installed
Automatically installed: no
Version: 0.4.4-1
Priority: optional
Section: python
Maintainer: William Woodall <william@osrfoundation.org>
Architecture: all
Uncompressed Size: 885 k
Depends: python (>= 2.7), python (< 2.8), python:any (>= 2.7.1-0ubuntu2),
         python-catkin-pkg (>= 0.2.9), python-pkg-resources, python-yaml,
         python-osrf-pycommon, python-argparse, python-setuptools
Conflicts: python3-catkin-tools, python3-catkin-tools
Description: Command line tools for working with catkin.
 Provides command line tools for working with catkin.
```

> とりあえず治したければmerge-develを使うか、
> catkin-tools==0.3.Xを使えばいいかと思います。

普通にやって動くものが、普通にやって動かない時に、普通でない方法で動かそうとするのは陥りやすい罠ですね．

普通にやって普通に動くので、普通に戻す方向が正解です．

ということで、真面目にlogをみてみると、

エラーになっているのは
```
mv: cannot stat ‘/home/muku/catkin_test/devel/.private/libsiftfast/lib/siftfastpy.so’: No such file or directory
```
なので、python犯人説ですね．
ということで、先頭からみていくと、

```
-- Found PythonInterp: /home/muku/.pyenv/shims/python (found version "2.7.6")
-- Using PYTHON_EXECUTABLE: /home/muku/.pyenv/shims/python
```
早速嫌な予感がしてきました．僕のログを見ると
```
$ grep python log.txt  | grep -- --
[catkin_tools_prebuild:cmake] -- Found PythonInterp: /usr/bin/python (found version "2.7.6")
[catkin_tools_prebuild:cmake] -- Using PYTHON_EXECUTABLE: /usr/bin/python
[libsiftfast:cmake] -- Found PythonInterp: /usr/bin/python (found version "2.7.6")
[libsiftfast:cmake] -- Using PYTHON_EXECUTABLE: /usr/bin/python
[libsiftfast:make] cd libsiftfast_svn && svn revert --recursive . && patch -p0 -f -E < /tmp/aaa/src/jsk_3rdparty/3rdparty/libsiftfast/patches/01.fix_python_binding.patch
[libsiftfast:make] --   python
[libsiftfast:make] -- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.6") 
[libsiftfast:make] -- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
```
美しいですね．py.so が落ちているから、PythonLibsとかが怪しいね．
で、@mukuact のものに戻ると、
```
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
```
またか．
```
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
```
あれ、これでいいんだっけ？pyenv使っているつもりがいつのまにかaptになっているよ．バージョン同じだけど、へんじゃない．
```
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found version "2.7.6")
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
```

ここも、apt．その下をみていると、
```
-- Performing Test HAVE_ALL_PYTHON_HEADERS
-- Performing Test HAVE_ALL_PYTHON_HEADERS - Failed
-- failed to use boost python libraries, check if python-numpy is installed.
```
エラーが出ている．これで直ると思う．
package.xmlをみると、
```
 <build_depend>python-numpy</build_depend>
```
となっているから、結局、アドバイスとしては、コンパイルするなら
```
rosdep install --from-path . --ignore-src 
```
しましょう．でも、
```
sudo apt-get install ros-indigo-libsiftfast
```
で終わるんだから、これで良いと思う．

[log.txt](https://github.com/jsk-ros-pkg/jsk_3rdparty/files/963323/log.txt)

closed 2017-04-28 by k-okada

wkentaro, 2017-04-28:
なるほど、rosdepはしていたけど、pyenvが起動していたから、systemのnumpy libが見つからなかったということですね。

k-okada, 2017-04-28:
rosdep したら、apt-get install 走るんじゃないかな．だとすると、かならず、system
にnumpyがはいるはず．ということで、rosdepしていないに一票．

wkentaro, 2017-04-28:
pyenvは使っていないので推測ですが、仮想環境を作るとsystemのlibを見ないということもあるのではないかと思うんですよね。
ここで https://github.com/jsk-ros-pkg/jsk_3rdparty/issues/93#issue-224027936 rosdepはしていると書いてるのと`python-numpy`が入っていないというのは中々ない環境かなと思うので。。

k-okada, 2017-04-28:
```
rosdep install --from-paths src --ignore-src -r -n -y
```
見逃していました．なるほど、たしかに、
```
    check_cxx_source_runs("
    #include <Python.h>
    #include <boost/thread/thread.hpp>
    #include <boost/thread/mutex.hpp>
    #include <boost/thread/condition.hpp>
    #include <boost/shared_ptr.hpp>
    #define PY_ARRAY_UNIQUE_SYMBOL PyArrayHandle
    #include <boost/python.hpp>
    #include <pyconfig.h>
    #include <numpy/arrayobject.h>
    int main()
    {
      return 0;
    }"
    HAVE_ALL_PYTHON_HEADERS)

    if( HAVE_ALL_PYTHON_HEADERS )
      message(STATUS "python and boost-python found")
      include_directories(${PYTHON_INCLUDE_PATH} ${Boost_INCLUDE_DIRS})
      link_directories(${Boost_LIBRARY_DIRS})
      add_library(siftfastpy SHARED siftfastpy.cpp libsiftfast.cpp)
      # stdc++ has to be included before opengl libraries due to some ATI bug (h
ttp://wiki.fifengine.de/Segfault_in_cxa_allocate_exception#Workaround)
      target_link_libraries(siftfastpy ${STDC_LIBRARY} ${PYTHON_LIBRARIES} ${Boo
st_PYTHON_LIBRARY})

      set_target_properties(siftfastpy PROPERTIES PREFIX "")
      if( WIN32 )
        set_target_properties(siftfastpy PROPERTIES SUFFIX ".pyd")
      endif()
      set(BUILD_SIFTFASTPY 1)
      install(TARGETS siftfastpy DESTINATION lib PERMISSIONS
        OWNER_READ OWNER_WRITE OWNER_EXECUTE
        GROUP_READ GROUP_EXECUTE
        WORLD_READ WORLD_EXECUTE)
    else()
      message(STATUS "failed to use boost python libraries, check if python-numpy is installed.")
    endif()
```
なので、結構難しそうだ．

iory, 2018-08-20:
遅くなりましたが、これは```pip```でインストールした場合と```apt```でインストールした時のpathが違うことで起こるのではないでしょうか？

numpyは
pipだと
```
/usr/local/lib/python2.7/dist-packages/numpy/core/include/numpy/
```
aptだと
```
/usr/lib/python2.7/dist-packages/numpy/core/include/numpy/
```
にインストールされています。

iory, 2018-08-20:
#151　にてnumpyのincludepathを追加するpatchを作成しました。

k-okada, 2018-08-28:
apt-get ros-kinetic-libsiftfastしたら，python-numpyが入るべきでは？入らないなら入れるように．入っていれば，パスは必要ないのでは？

iory, 2018-08-28:
> apt-get ros-kinetic-libsiftfastしたら，python-numpyが入るべきでは？入らないなら入れるように．入っていれば，パスは必要ないのでは？

日本語の意味の確認ですが、pipでインストールする人が悪いという意味でしょうか？

k-okada, 2018-08-28:
pipでインストールしていても，apt-get instal
ros....libsiftfastしたらpython-numpyがはいるのでは？という意味です．
--
◉ Kei Okada

iory, 2018-08-28:
でもrosdepしてsystemのnumpyが入ってもbuildが通りませんでしたというのがこの上のほうの話ではなかったでしょうか？

k-okada, 2018-08-28:
pipのnumpyにパスをとおそたから動きました、というのは違うのでは？ということです
システムのパスは存在するんだから、それを使って何かやる方法があるはず。
というのと
pipで動きますは、全てのpython/numpy のバージョンで動きます、ということだから収集つかなるはず

iory, 2018-08-28:
そういう意味でしたか。

> システムのパスは存在するんだから、それを使って何かやる方法があるはず。

ということなので、システムのパスでbuildする方法を探します。

kochigami, 2019-10-15:
`catkin b libsiftfast -v`すると，
`failed to use boost python libraries, check if python-numpy is installed.`
`mv: cannot stat '/home/pepper/catkin_ws/devel/.private/libsiftfast/lib/siftfastpy.so': No such file or directory`
が出ます．
ubuntuをクリーンインストールしようかと思っているのですが，その前に他に試すことがありましたらご教示ください．
pipでnumpyを入れたと思います．（明日確認します）
コンパイル時のログを見るに，pyenvは使っていないと思います．
ubuntu16.04です．コンパイル2回目以降もエラーが出ます．

python-numpyのインストール先（下の長いログ）
しかし
```
pepper@pepper-thinkpad-t480s:/usr/lib/python2.7/dist-packages$ ls | grep numpy
```
は何も出力しない

```
pepper@pepper-thinkpad-t480s:~$ dpkg -L python-numpy
/.
/usr
/usr/lib
/usr/lib/python2.7
/usr/lib/python2.7/dist-packages
/usr/lib/python2.7/dist-packages/numpy-1.11.0.egg-info
/usr/lib/python2.7/dist-packages/numpy-1.11.0.egg-info/dependency_links.txt
/usr/lib/python2.7/dist-packages/numpy-1.11.0.egg-info/PKG-INFO
/usr/lib/python2.7/dist-packages/numpy-1.11.0.egg-info/top_level.txt
/usr/lib/python2.7/dist-packages/numpy
/usr/lib/python2.7/dist-packages/numpy/lib
/usr/lib/python2.7/dist-packages/numpy/lib/shape_base.py
/usr/lib/python2.7/dist-packages/numpy/lib/ufunclike.py
/usr/lib/python2.7/dist-packages/numpy/lib/arrayterator.py
/usr/lib/python2.7/dist-packages/numpy/lib/nanfunctions.py
/usr/lib/python2.7/dist-packages/numpy/lib/stride_tricks.py
/usr/lib/python2.7/dist-packages/numpy/lib/_iotools.py
/usr/lib/python2.7/dist-packages/numpy/lib/info.py
/usr/lib/python2.7/dist-packages/numpy/lib/type_check.py
/usr/lib/python2.7/dist-packages/numpy/lib/recfunctions.py
/usr/lib/python2.7/dist-packages/numpy/lib/user_array.py
/usr/lib/python2.7/dist-packages/numpy/lib/utils.py
/usr/lib/python2.7/dist-packages/numpy/lib/setup.py
/usr/lib/python2.7/dist-packages/numpy/lib/scimath.py
/usr/lib/python2.7/dist-packages/numpy/lib/__init__.py
/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py
/usr/lib/python2.7/dist-packages/numpy/lib/twodim_base.py
/usr/lib/python2.7/dist-packages/numpy/lib/arraypad.py
/usr/lib/python2.7/dist-packages/numpy/lib/polynomial.py
/usr/lib/python2.7/dist-packages/numpy/lib/function_base.py
/usr/lib/python2.7/dist-packages/numpy/lib/arraysetops.py
/usr/lib/python2.7/dist-packages/numpy/lib/_datasource.py
/usr/lib/python2.7/dist-packages/numpy/lib/format.py
/usr/lib/python2.7/dist-packages/numpy/lib/index_tricks.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_index_tricks.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_arrayterator.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_financial.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test__version.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_io.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test__datasource.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_stride_tricks.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_format.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_utils.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/test_shape_base.py
/usr/lib/python2.7/dist-packages/numpy/lib/tests/data
/usr/lib/python2.7/dist-packages/numpy/lib/tests/data/py3-objarr.npz
/usr/lib/python2.7/dist-packages/numpy/lib/tests/data/py2-objarr.npy
/usr/lib/python2.7/dist-packages/numpy/lib/tests/data/py2-objarr.npz
/usr/lib/python2.7/dist-packages/numpy/lib/tests/data/win64python2.npy
/usr/lib/python2.7/dist-packages/numpy/lib/tests/data/python3.npy
/usr/lib/python2.7/dist-packages/numpy/lib/tests/data/py3-objarr.npy
[... 484 lines cut ...]
mkdir -p /home/pepper/catkin_ws/devel/.private/libsiftfast/lib/python2.7/dist-packages && mv /home/pepper/catkin_ws/devel/.private/libsiftfast/lib/siftfastpy.so /home/pepper/catkin_ws/devel/.private/libsiftfast/lib/python2.7/dist-packages/siftfastpy.so
WARN, found multiple boost versions '[(1, 58, 0, '/usr', '/usr/include', True, True), (1, 58, 0, '/usr', '/usr/include', True, True)]', using latestWARN, found multiple boost versions '[(1, 58, 0, '/usr', '/usr/include', True, True), (1, 58, 0, '/usr', '/usr/include', True, True)]', using latestmv: cannot stat '/home/pepper/catkin_ws/devel/.private/libsiftfast/lib/siftfastpy.so': No such file or directory
/home/pepper/catkin_ws/src/jsk_3rdparty/3rdparty/libsiftfast/Makefile:29: recipe for target 'libsiftfast' failed
make[3]: Leaving directory '/home/pepper/catkin_ws/build/libsiftfast'
make[3]: *** [libsiftfast] Error 1
CMakeFiles/libsiftfast.dir/build.make:60: recipe for target '/home/pepper/catkin_ws/devel/.private/libsiftfast/lib/libsiftfast.so' failed
make[2]: *** [/home/pepper/catkin_ws/devel/.private/libsiftfast/lib/libsiftfast.so] Error 2
make[2]: *** Deleting file '/home/pepper/catkin_ws/devel/.private/libsiftfast/lib/libsiftfast.so'
make[2]: Leaving directory '/home/pepper/catkin_ws/build/libsiftfast'
CMakeFiles/Makefile2:259: recipe for target 'CMakeFiles/libsiftfast.dir/all' failed
make[1]: Leaving directory '/home/pepper/catkin_ws/build/libsiftfast'
make[1]: *** [CMakeFiles/libsiftfast.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
cd /home/pepper/catkin_ws/build/libsiftfast; catkin build --get-env libsiftfast | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << libsiftfast:make           [ Exited with code 2 ]                
Failed    <<< libsiftfast                [ 1.8 seconds ]                       
[build] Summary: 0 of 1 packages succeeded.                                    
[build]   Ignored:   123 packages were skipped or are blacklisted.             
[build]   Warnings:  None.                                                     
[build]   Abandoned: None.                                                     
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 2.5 seconds total.      
```

k-okada, 2019-10-15:
クリーンインストールもpip install も やらないほうがいいとおもいますが、メッセージを見るに
 libboost-python-dev
か何かがないかおかしいようにみえます。
catkin build libsiftfast -vi —make-args VERBOSE=1
したらもっとエラーメッセージが出るかと思います