jsk-ros-pkg/jsk_3rdparty #98: rostwitter get_access_token.py not found

furushchev, 2017-05-15:
I'm re-configuring PR2 and I must re-issue access token for twitter.
when I execute `rosrun rostwitter tweet.py`, an error occured.

```
[ERROR] [WallTime: 1494841351.862268] "/var/lib/robot/twitter_account_pr2jsk.yaml" not found
[ERROR] [WallTime: 1494841351.863620] $ rosrun rostwitter get_access_token.py
[ERROR] [WallTime: 1494841351.864082] cat /var/lib/robot//var/lib/robot/twitter_account_pr2jsk.yaml <<EOF
[ERROR] [WallTime: 1494841351.864569] CKEY: xxx
[ERROR] [WallTime: 1494841351.865008] CSECRET: xxx
[ERROR] [WallTime: 1494841351.865596] AKEY: xxx
[ERROR] [WallTime: 1494841351.865991] ASECRET:
```
but I coud not find `get_access_token.py`. How can we use this?
CC: @k-okada 
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/rostwitter/scripts/tweet.py#L129

k-okada, 2017-05-15:
https://github.com/bear/python-twitter/blob/master/get_access_token.py ?

--
◉ Kei Okada

k-okada, 2017-05-15:
but you can copy twitter_account_pr2jsk.yaml from pr1040
--
◉ Kei Okada

2017-05-15 19:00 GMT+09:00 Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>:
> https://github.com/bear/python-twitter/blob/master/get_access_token.py ?
>
>
> --
> ◉ Kei Okada
>

furushchev, 2017-05-19:
@k-okada Thanks for information and yes I copied from pr1040 and now it works fine!

PR #100 (2017-05-24): rostwitter: fix error message when access token is not found
please review @furushchev, close #98
closed 2017-05-19 by furushchev