jsk-ros-pkg/jsk_3rdparty #103: Request: stop cron_for_github and use cron function by travis

furushchev, 2017-06-27:
This is not high priority.
All forked repositories from here must have many branches:

```bash
remotes/origin/cron_for_github/02bc619a-a7e5-4046-8621-67b1d7e1b0b2
  remotes/origin/cron_for_github/02e191ed-34b3-41b0-89d4-03b9ab773fc4
  remotes/origin/cron_for_github/1536df62-91e1-4848-82eb-9a2bcfd84a5c
  remotes/origin/cron_for_github/165a5c88-a739-47d4-8e35-ad696cc07002
  remotes/origin/cron_for_github/2c3497af-5ecd-4012-8473-9c9124707dc1
  remotes/origin/cron_for_github/3f0ceaa1-3eea-468e-80c3-c4dd306806da
  remotes/origin/cron_for_github/3fde9d17-6222-4d1d-bb05-b29d99c9b674
  remotes/origin/cron_for_github/4d31ae4c-74e6-4725-a91f-ae47628defc5
  remotes/origin/cron_for_github/4e33e2d9-9ea6-405f-876c-702139770d62
  remotes/origin/cron_for_github/6701a916-eb61-4e16-81d6-910fc8db23bd
  remotes/origin/cron_for_github/674faaf1-b82c-4674-a30a-b2e5e2955938
  remotes/origin/cron_for_github/759471c7-0882-47dc-9200-e9f556e9c334
  remotes/origin/cron_for_github/78268f0e-6d65-4ae2-ade1-e73762f64f4a
  remotes/origin/cron_for_github/8323e2e3-8d99-4fc2-a740-04ac51035db5
  remotes/origin/cron_for_github/83fed874-b88b-449b-ae74-cf5543acff4a
  remotes/origin/cron_for_github/890080b4-1032-4adb-a639-e2a39a336873
  remotes/origin/cron_for_github/959ec0e4-f94e-4560-bc74-842918108b58
  remotes/origin/cron_for_github/96ea09d2-4dab-44fe-bff0-9fa19ea9f164
  remotes/origin/cron_for_github/9a3f5a88-7c4b-47c7-a300-081340c51ae9
  remotes/origin/cron_for_github/a015dbf5-f26c-4841-a0c1-e828971d33c5
  remotes/origin/cron_for_github/b22f6450-7e4d-4a5b-8abb-727f0abd0cdb
  remotes/origin/cron_for_github/b2fb03a7-eddb-413a-9852-d467430efd8d
  remotes/origin/cron_for_github/b34c4d38-d880-4fcf-8b68-f01fdd113ad6
  remotes/origin/cron_for_github/b8c3a8f4-a60c-43d2-9129-8ad70c9ddc72
  remotes/origin/cron_for_github/bdcc0ef9-0e55-4a29-b827-f9d5994ed819
  remotes/origin/cron_for_github/c10fc506-0540-4466-854a-222faf7c3eb0
  remotes/origin/cron_for_github/ca7d5f69-6862-4cd2-bf5d-d41802dbb768
  remotes/origin/cron_for_github/cf8a721a-542e-4472-bb1c-90c851b36086
  remotes/origin/cron_for_github/e3e247da-3cbb-4d1b-9e0c-f96e1a9cefd2
  remotes/origin/cron_for_github/e7625d49-3efd-499e-bae8-de72507d8de6
  remotes/origin/cron_for_github/ea662263-f357-40dc-a08a-4a1e21576195
  remotes/origin/cron_for_github/ef18dfb8-d618-4985-9aa1-b170e7b6060c
  remotes/origin/cron_for_github/f57af4b2-46e4-44bf-b18f-b650f6fc8140
  remotes/origin/cron_for_github/f8ceb3b6-eebe-449a-8ab0-ea54eb2617d1
  remotes/origin/cron_for_github/fe29fecc-cc9f-475e-a61e-6da6e994c3fb
```

Travis now officially provides cron testing function. Please consider to use it to keep clean branches.

furushchev, 2017-06-29:
@k-okada Could you let me change the settings of this repository in a mean time?

k-okada, 2017-06-30:
I think I have removed unused branches and enabled   Travis cron setting.

furushchev, 2017-06-30:
@k-okada I did not notice that! Confirmed that all cron branches are now removed. Thank you!

closed 2017-06-30 by furushchev