jsk-ros-pkg/jsk_3rdparty #108: Build failure `julius` on buildfirm

furushchev, 2017-07-03:
I found that we have released new version of packages in this repository, but julius provided via apt  is still `2.0.20`.

```bash
$ sudo apt show ros-indigo-julius
[sudo] password for furushchev: 
Package: ros-indigo-julius
Version: 2.0.20-0trusty-20170509-112617-0800
Maintainer: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Installed-Size: 2,493 kB
Depends: libc6 (>= 2.15), zlib1g (>= 1:1.1.4)
Homepage: http://ros.org/wiki/julius
Priority: extra
Section: misc
Download-Size: 370 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros-shadow-fixed/ubuntu/ trusty/main amd64 Packages
Description: julius: Open-Source Large Vocabulary CSR Engine (http://julius.sourceforge.jp/index.php)

$ sudo apt show ros-indigo-julius-ros
Package: ros-indigo-julius-ros
Version: 2.1.0-0trusty-20170701-225619-0800
Maintainer: Yuki Furuta <furushchev@jsk.imi.i.u-tokyo.ac.jp>
Installed-Size: 148 kB
Depends: nkf, python-lxml, python-rospkg, ros-indigo-audio-capture, ros-indigo-audio-common-msgs, ros-indigo-julius, ros-indigo-rospy, ros-indigo-sound-play, ros-indigo-speech-recognition-msgs, ros-indigo-std-srvs
Priority: extra
Section: misc
Download-Size: 19.2 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros-shadow-fixed/ubuntu/ trusty/main amd64 Packages
Description: The julius_ros package
```

I found the build step on buildfirm fails:

http://build.ros.org/view/Isrc_uT/job/Ibin_uT64__julius__ubuntu_trusty_amd64__binary/11/consoleFull

```bash
23:51:36 configure: running /bin/bash ./configure --disable-option-checking '--prefix=/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu/devel/share/julius'  '--enable-decoder-vad' 'CFLAGS=-g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security' 'LDFLAGS=' 'CPPFLAGS=-D_FORTIFY_SOURCE=2' --cache-file=/dev/null --srcdir=.
23:51:36 configure: warning: CFLAGS=-g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security: invalid host type
23:51:36 configure: warning: LDFLAGS=: invalid host type
23:51:36 configure: error: can only configure for one host and one target at a time
23:51:36 configure: error: ./configure failed for mkgshmm
23:51:36 make[6]: Entering directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu/build/julius-4.4.2'
23:51:36 make[6]: *** [libsent] Error 1
23:51:36 make[6]: Leaving directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu/build/julius-4.4.2'
23:51:36 make[6]: Entering directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu/build/julius-4.4.2'
23:51:36 make[6]: *** [libsent] Error 1
23:51:36 make[6]: Leaving directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu/build/julius-4.4.2'
23:51:36 make[5]: *** [installed.julius] Error 2
23:51:36 make[5]: Leaving directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu'
23:51:36 make[4]: *** [julius_installed] Error 2
23:51:36 make[3]: *** [CMakeFiles/all_installed.dir/all] Error 2
23:51:36 make[2]: *** [all] Error 2
23:51:36 make[4]: Leaving directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu'
23:51:36 make[3]: Leaving directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu'
23:51:36 make[2]: Leaving directory `/tmp/binarydeb/ros-indigo-julius-2.1.0/obj-x86_64-linux-gnu'
23:51:36 dh_auto_build: make -j1 returned exit code 2
23:51:36 make[1]: *** [override_dh_auto_build] Error 2
23:51:36 make[1]: Leaving directory `/tmp/binarydeb/ros-indigo-julius-2.1.0'
23:51:36 make: *** [build] Error 2
23:51:36 dpkg-buildpackage: error: debian/rules build gave error exit status 2
23:51:36 E: Building failed
23:51:36 Traceback (most recent call last):
23:51:36   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
23:51:36     subprocess.check_call(cmd, cwd=source_dir)
23:51:36   File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
23:51:36     raise CalledProcessError(retcode, cmd)
23:51:36 subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-indigo-julius']' returned non-zero exit status 1
```

k-okada, 2017-07-03:
> _ubuntu_trusty_amd64__binary/11/consoleFull

maybe it is not our problem, so let's wait fo moment ->
https://groups.google.com/forum/#!forum/ros-buildfarm-indigo

furushchev, 2017-07-03:
@k-okada OK! Thanks!

FYI: I locally runned:

```bash
roscd julius
bloom-generate rosdebian --os-name ubuntu --os-version trusty --ros-distro indigo
fakeroot debian/rules binary
```

and the same error is reproduced.

k-okada, 2017-07-03:
2017-07-03 17:06 GMT+09:00 Furushchev <notifications@github.com>:

> bloom-generate rosdebian --os-name ubuntu --os-version trusty --ros-distro
> indigo fakeroot debian/rules binary

which version of bloom-generate are you using?
```
$ aptitude show python-bloom
Package: python-bloom
State: installed
Automatically installed: no
Version: 0.5.26-1
Priority: optional
Section: python
Maintainer: William Woodall <william@osrfoundation.org>
Architecture: all
Uncompressed Size: 466 k
Depends: python (>= 2.7), python (< 2.8), python:any (>= 2.7.1-0ubuntu2),
python-catkin-pkg (>= 0.2.2),
         python-pkg-resources, python-empy, python-dateutil, python-yaml,
python-rosdep (>= 0.10.25), python-rosdistro (>=
         0.4.0), python-vcstools (>= 0.1.22), python-argparse,
python-setuptools
Conflicts: python3-bloom, python3-bloom
Description: Bloom is a release automation tool.
 Bloom provides tools for releasing software on top of a git repository and
leverages tools and patterns from
 git-buildpackage. Additionally, bloom leverages meta and build information
from catkin (https://github.com/ros/catkin) to
 automate release branching and the generation of platform specific source
packages, like debian's src-debs.

```

--
◉ Kei Okada

furushchev, 2017-07-03:
@k-okada Oh, I installed via `pip` not via `apt`.

```
$ pip freeze | grep bloom
bloom==0.5.26
```

It seems the same version anyway, I'll try after reinstall via `apt`.

furushchev, 2017-07-03:
Still same error occurs if `bloom` is installed via `apt`.
Here is the full log:
https://pastebin.com/qTbcSpzW

k-okada, 2017-07-03:
```
k-okada@p40-yoga:/tmp/jsk_3rdparty/julius_ros$ fakeroot debian/rules binary
dh  binary
   dh_testdir
Unknown option: buildsystem
dh_testdir: warning: ignored unknown options in DH_OPTIONS
   debian/rules override_dh_auto_configure
make[1]: Entering directory `/tmp/jsk_3rdparty/julius_ros'
# In case we're installing to a non-standard location, look for a setup.sh
# in the install tree that was dropped by catkin, and source it.  It will
# set things like CMAKE_PREFIX_PATH, PKG_CONFIG_PATH, and PYTHONPATH.
if [ -f "/opt/ros/indigo/setup.sh" ]; then . "/opt/ros/indigo/setup.sh"; fi && \
	dh_auto_configure -- \
		-DCATKIN_BUILD_BINARY_PACKAGE="1" \
		-DCMAKE_INSTALL_PREFIX="/opt/ros/indigo" \
		-DCMAKE_PREFIX_PATH="/opt/ros/indigo"
	mkdir -p obj-x86_64-linux-gnu
	cd obj-x86_64-linux-gnu
	cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCATKIN_BUILD_BINARY_PACKAGE=1 -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_PREFIX_PATH=/opt/ros/indigo
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Using CATKIN_DEVEL_PREFIX: /tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() when building binary package
-- Using CATKIN_TEST_RESULTS_DIR: /tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.6.19
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu
	cd /tmp/jsk_3rdparty/julius_ros
make[1]: Leaving directory `/tmp/jsk_3rdparty/julius_ros'
   debian/rules override_dh_auto_build
make[1]: Entering directory `/tmp/jsk_3rdparty/julius_ros'
# In case we're installing to a non-standard location, look for a setup.sh
# in the install tree that was dropped by catkin, and source it.  It will
# set things like CMAKE_PREFIX_PATH, PKG_CONFIG_PATH, and PYTHONPATH.
if [ -f "/opt/ros/indigo/setup.sh" ]; then . "/opt/ros/indigo/setup.sh"; fi && \
	dh_auto_build
	cd obj-x86_64-linux-gnu
	make -j1
make[2]: Entering directory `/tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu'
/usr/bin/cmake -H/tmp/jsk_3rdparty/julius_ros -B/tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu/CMakeFiles /tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[3]: Entering directory `/tmp/jsk_3rdparty/julius_ros/obj-x86_64-linux-gnu'
[... 237 lines cut ...]
dh_shlibdeps: warning: ignored unknown options in DH_OPTIONS
make[1]: Leaving directory `/tmp/jsk_3rdparty/julius_ros'
   dh_installdeb
Unknown option: buildsystem
dh_installdeb: warning: ignored unknown options in DH_OPTIONS
   dh_gencontrol
Unknown option: buildsystem
dh_gencontrol: warning: ignored unknown options in DH_OPTIONS
	echo misc:Depends= >> debian/ros-indigo-julius-ros.substvars
	dpkg-gencontrol -ldebian/changelog -Tdebian/ros-indigo-julius-ros.substvars -Pdebian/ros-indigo-julius-ros
dpkg-gencontrol: warning: Depends field of package ros-indigo-julius-ros: unknown substitution variable ${shlibs:Depends}
	chmod 644 debian/ros-indigo-julius-ros/DEBIAN/control
	chown 0:0 debian/ros-indigo-julius-ros/DEBIAN/control
   dh_md5sums
Unknown option: buildsystem
dh_md5sums: warning: ignored unknown options in DH_OPTIONS
	(cd debian/ros-indigo-julius-ros >/dev/null ; find . -type f  ! -regex './DEBIAN/.*' -printf '%P\0' | LC_ALL=C sort -z | xargs -r0 md5sum > DEBIAN/md5sums) >/dev/null
	chmod 644 debian/ros-indigo-julius-ros/DEBIAN/md5sums
	chown 0:0 debian/ros-indigo-julius-ros/DEBIAN/md5sums
   dh_builddeb
Unknown option: buildsystem
dh_builddeb: warning: ignored unknown options in DH_OPTIONS
	dpkg-deb --build debian/ros-indigo-julius-ros ..
dpkg-deb: building package `ros-indigo-julius-ros' in `../ros-indigo-julius-ros_2.1.0-0trusty_amd64.deb'.
```

k-okada, 2017-07-03:
please compare your generated `debian` directory and https://github.com/tork-a/jsk_3rdparty-release/tree/debian/indigo/trusty/julius_ros, ONLY IF you'd like to debug.

furushchev, 2017-07-03:
@k-okada Thanks for reply! Your log is for `julius_ros`, not `julius` itself, I think.

k-okada, 2017-07-03:
isee may be

k-okada@p40-yoga:/tmp/jsk_3rdparty-release$ git diff
diff --git a/Makefile.julius b/Makefile.julius
index 03b9b6c..dbc6e2d 100644
--- a/Makefile.julius
+++ b/Makefile.julius
@@ -20,7 +20,7 @@ LIBSENT_OPTION =
 LIBJULIUS_OPTION = "--enable-decoder-vad"

 installed.julius: $(SOURCE_DIR)/unpacked
-   (cd $(SOURCE_DIR); ./configure --prefix=$(INSTALL_DIR)
$(LIBSENT_OPTION) $(LIBJULIUS_OPTION); make; make install)
+ (cd $(SOURCE_DIR); unset LDFLAG; unset CFLAGS; unset CPPFLAGS;
./configure --prefix=$(INSTALL_DIR) $(LIBSENT_OPTION) $(LIBJULIUS_OPTION);
make; make install)
        touch installed.julius

k-okada, 2017-07-03:
Running fakeroot will build deb file, so please install that file and
confirm if everything works fine

furushchev, 2017-07-03:
@k-okada I checked and confirmed it works.

```bash
furushchev@furushchev-ThinkPad-T460s:/opt/ros/indigo/share/julius $ ls
bin  cmake  conf  include  lib  model  package.xml  SampleGrammars  SampleGrammars_en
```

```bash
roscd julius_ros/data
$ rosrun julius generate udon
Stat: init_voca: read 14 words
Reading in term file (optional)...done
5 categories, 14 words
DFA has 6 nodes and 6 arcs
----- 
 <s> 8 udon please </s>
 <s> udon please </s>
 <s> 9 udon please </s>
 <s> 0 udon please </s>
 <s> 6 udon please </s>
 <s> 5 udon please </s>
 <s> 3 udon please </s>
 <s> 2 udon please </s>
 <s> 1 udon please </s>
 <s> 4 udon please </s>
```

furushchev, 2017-07-11:
@k-okada Thanks now available on `apt`.

```bash
$ sudo apt show ros-indigo-julius
Package: ros-indigo-julius
Status: install ok installed
Priority: extra
Section: misc
Installed-Size: 851 MB
Maintainer: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Version: 2.1.0-0trusty
Depends: libasound2 (>= 1.0.16), libc6 (>= 2.15), libpulse0 (>= 1:0.99.1), libreadline6 (>= 6.0), libsdl2-2.0-0 (>= 2.0.0), libsndfile1 (>= 1.0.20), zlib1g (>= 1:1.1.4)
Homepage: http://ros.org/wiki/julius
Download-Size: unknown
APT-Manual-Installed: yes
APT-Sources: /var/lib/dpkg/status
Description: julius: Open-Source Large Vocabulary CSR Engine (http://julius.sourceforge.jp/index.php)
```

closed 2017-07-11 by furushchev