jsk-ros-pkg/jsk_3rdparty #113: downward really needs g++-static?

Alessandro-Barbieri, 2017-07-06:
Since there is no `g++-static` in gentoo I'm asking if it's really needed or if can be replaced with standard `g++`

k-okada, 2017-07-06:
humm, no idea.
Can you try building on gentoo environment and see if
https://github.com/jsk-ros-pkg/jsk_3rdparty/tree/master/3rdparty/downward/test
works.

Alessandro-Barbieri, 2017-07-07:
Seems that the `-static` stuff is Fedora only, in ubuntu arch and debian is the normal gcc

k-okada, 2017-07-07:
Thank you for your investigation, should we change to gcc? Do you want to
remove gcc-static from rodep rule?

Alessandro-Barbieri, 2017-07-07:
current entry in rosdep is 

```
g++-static:
  arch: [gcc]
  debian: [g++]
  fedora: [gcc-c++, glibc-devel, glibc-static, libstdc++-devel, libstdc++-static]
  ubuntu: [g++]
```

> should we change to gcc?

changing to gcc would be great

k-okada, 2017-07-08:
thanks, I have changed to gcc,
https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/115