jsk-ros-pkg/jsk_3rdparty #118: julius ros package is 300+ MB

tfoote, 2017-08-12:
Auditing our bandwidth and storage of ROS debians julius was noticed to be a large outlier

```
372K	ros-kinetic-julius/ros-kinetic-julius_2.0.19-0wily-20170308-192239-0800_i386.deb
368K	ros-kinetic-julius/ros-kinetic-julius_2.0.19-0wily-20170309-024528-0800_amd64.deb
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.0.19-0wily.debian.tar.xz
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.0.19-0wily.dsc
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.0.19.orig.tar.gz
334M	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0jessie-20170716-092221-0800_amd64.deb
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0jessie.debian.tar.xz
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0jessie.dsc
334M	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0xenial-20170716-090619-0800_amd64.deb
333M	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0xenial-20170716-090748-0800_i386.deb
334M	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0xenial-20170716-093027-0800_armhf.deb
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0xenial.debian.tar.xz
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.1.4-0xenial.dsc
4.0K	ros-kinetic-julius/ros-kinetic-julius_2.1.4.orig.tar.gz
1.4G	total
```

This is an order of magnitude larger than any other packages in desktop full: https://gist.github.com/tfoote/c29f0230c1a6a49c6c83c5b62f712573 including pr2_description with large meshes. 

This causes it to become approximately 20% of the content in the kinetic repository
```
du -hcs ros-kinetic* | sort -h | tail -n 12
103M	ros-kinetic-jsk-pcl-ros
104M	ros-kinetic-ros-peerjs
114M	ros-kinetic-cob-mimic
119M	ros-kinetic-visp-tracker
141M	ros-kinetic-parrot-arsdk
175M	ros-kinetic-jsk-perception
183M	ros-kinetic-visp-auto-tracker
204M	ros-kinetic-opencv3
231M	ros-kinetic-moveit-resources
255M	ros-kinetic-pr2-description
1.4G	ros-kinetic-julius
5.6G	total
```

It looks like there's a lot of data in share
```
root@815dd4a2e78d:/var/cache/apt# dpkg -L ros-kinetic-julius | xargs -l1 du -hs | sort -h
du: cannot access '/./proc/31883/task/31883/fd/4': No such file or directory
du: cannot access '/./proc/31883/task/31883/fdinfo/4': No such file or directory
du: cannot access '/./proc/31883/fd/3': No such file or directory
du: cannot access '/./proc/31883/fdinfo/3': No such file or directory
4.0K	/opt/ros/kinetic/lib/pkgconfig/julius.pc
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/00readme.txt
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/00readme.txt
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/Test.bat
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/Test.sh
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/attendant.dfa
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/attendant.dict
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/attendant.grammar
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/attendant.term
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/attendant.voca
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/attendant.xml
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/attendant/testmic.jconf
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/datetime/00readme.txt
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/datetime/Test.bat
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/datetime/Test.sh
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/datetime/datetime.dfa
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/datetime/datetime.grammar
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/datetime/datetime.term
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/datetime/testmic.jconf
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/digit/00readme.txt
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/digit/Test.bat
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/digit/Test.sh
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/digit/digit.dfa
4.0K	/opt/ros/kinetic/share/julius/SampleGrammars/digit/digit.dict
[... 223 lines cut ...]
17M	/opt/ros/kinetic/share/julius/model/dnn/W_l5_f4.npy
17M	/opt/ros/kinetic/share/julius/model/dnn/W_l6_f4.npy
17M	/opt/ros/kinetic/share/julius/model/dnn/W_l7_f4.npy
36M	/opt/ros/kinetic/share/julius/model/phone_m/jnas-tri-3k16-gid.hmmdefs
39M	/opt/ros/kinetic/share/julius/model/dnn/W_output_f4.npy
53M	/opt/ros/kinetic/share/julius/model/dnn/binhmm.SID
55M	/opt/ros/kinetic/share/julius/model/phone_m
82M	/opt/ros/kinetic/share/julius/model/lang_m/bccwj.60k.tri.arpa.gz
92M	/opt/ros/kinetic/share/julius/model/lang_m/bccwj.60k.bingram
113M	/opt/ros/kinetic/share/julius/model/dnn/logicalTri
122M	/usr/share/doc
130M	/opt/ros/kinetic/lib
179M	/opt/ros/kinetic/share/julius/model/lang_m
264M	/opt/ros/kinetic/share/julius/model/dnn/logicalTri.bin
418M	/usr/share
574M	/opt/ros/kinetic/share/julius/model/dnn
806M	/opt/ros/kinetic/share/julius/model
812M	/opt/ros/kinetic/share/julius
854M	/opt/ros/kinetic/share
1001M	/opt
1001M	/opt/ros
1001M	/opt/ros/kinetic
3.0G	/usr
4.9G	/.
```

k-okada, 2017-08-12:
@Furushchev any idea?

furushchev, 2017-08-12:
It's almost the learned DNN weights used for speech recognition. 
Hmm, installing them to user space after apt install? I can write a script to do so, but it may occur unexpected behavior since Julius executable is very fragile.

k-okada, 2017-08-13:
how about download model files after install ?
https://github.com/tork-a/darknet-release/blob/master/patches/debian/preinst
?

try
```
git clone https://github.com/tork-a/jsk_3rdparty-release -b git checkout debian/indigo/trusty/julius
cd jsk_3rdparty-release
# update patches directory
dpkg-buildpackage -rfakeroot -uc -b
```

to create deb files on your local environment and check if this works.

furushchev, 2017-08-18:
@k-okada Thanks for advice!
I created `preinst` and confirmed that it works, but I'm not sure where to put in the repository.
https://gist.github.com/furushchev/1b9e5e65ba2b9a1473e7cf484227f601

k-okada, 2017-08-24:
ok, I have fixed release repo, we'll have small deb package from next
release.

https://github.com/tork-a/jsk_3rdparty-release/tree/release/kinetic/julius
https://github.com/tork-a/jsk_3rdparty-release/tree/release/jade/julius
https://github.com/tork-a/jsk_3rdparty-release/tree/release/indigo/julius

furushchev, 2017-12-05:
I think this issue was already solved and it's ok to be closed.

closed 2017-12-06 by k-okada