jsk-ros-pkg/jsk_3rdparty #132: julius_ros should depends on rsync

furushchev, 2018-03-16:
Since https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/112, `rsync` is required to build `julius_ros` package and it is registered in `build_depend`, but it looks that debian package `ros-indigo-julius-ros` still does not depend on `rsync`.

```bash
$ sudo apt-get install ros-indigo-julius-ros
(...many logs...)
/var/lib/dpkg/tmp.ci/preinst: line 59: rsync: command not found
dpkg: error processing archive /var/cache/apt/archives/ros-indigo-julius_2.1.6-0trusty-20171122-203936-0800_amd64.deb (--unpack):
 subprocess new pre-installation script returned error exit status 127
```

I'm using `ros-shadow-fixed`.

```bash
$ sudo apt show ros-indigo-julius-ros
Package: ros-indigo-julius-ros
Version: 2.1.6-0trusty-20171218-092711-0800
Maintainer: Yuki Furuta <furushchev@jsk.imi.i.u-tokyo.ac.jp>
Installed-Size: 148 kB
Depends: julius-voxforge, nkf, python-lxml, python-rospkg, ros-indigo-audio-capture, ros-indigo-audio-common-msgs, ros-indigo-julius, ros-indigo-rospy, ros-indigo-sound-play, ros-indigo-speech-recognition-msgs, ros-indigo-std-srvs
Priority: extra
Section: misc
Download-Size: 19.2 kB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros-shadow-fixed/ubuntu/ trusty/main amd64 Packages
Description: The julius_ros package
```

@k-okada Any idea?

furushchev, 2018-03-16:
```bash
2018-03-16 05:10:24 (17.0 MB/s) - â€˜/tmp/tmp.O9jnuxz692/dictation-kit-v4.4.zipâ€™ saved [426837341/426837341]

/var/lib/dpkg/tmp.ci/preinst: line 37: unzip: command not found

Extracted archive /tmp/tmp.O9jnuxz692/dictation-kit-v4.4.zip, but /tmp/tmp.O9jnuxz692/dictation-kit-v4.4
```

`unzip` is also required.

k-okada, 2018-03-16:
we need add them to run_depends of `julius` (not `juilus_ros` ...)
https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/112/files

closed 2018-03-19 by k-okada