jsk-ros-pkg/jsk_3rdparty #137: dependency on wget missing for julius

tfoote, 2018-04-16:
Like #132 wget is used in the preinst, so it needs to be a run dependency too.

Here's the console output of me debugging: https://gist.github.com/tfoote/a6d17f8549e3fc6a669410ba1798791a

I also ran into #133 which I think is fixed in the testing repo, but I don't see the wget dependency added.

Also following up on [this discourse thread](https://discourse.ros.org/t/ros-kinetic-julius-error/4484?u=tfoote)

furushchev, 2018-04-16:
@tfoote Thank you for reporting! Created PR to add `wget` as a dependency.

closed 2018-04-17 by k-okada

k-okada, 2018-04-17:
sorry for trouble, 2.1.8 fixed this issue, https://github.com/ros/rosdistro/pull/17483
https://gist.github.com/k-okada/6661d0da4b029f0c55928ede9c746a88

PR #138 (2018-04-17): julius: add wget to dependencies
Closes #137