jsk-ros-pkg/jsk_3rdparty #146: ros_speech_recognition  performs poor with noise

auvilink, 2018-07-08:
Any chance adding noise suppression / reduction to ros_speech_recognition?

Thanks.

k-okada, 2018-07-09:
ros_speech_recognition is just ros wrapper for https://pypi.org/project/SpeechRecognition/, so if they support nose suppression/reduction  then we can use them from the package. 
@auvilink Did you tested some of https://github.com/Uberi/speech_recognition/blob/master/examples/calibrate_energy_threshold.py ? Does this code work with other engines? Your feedback will enable us to support these features in our package. 
Thanks

auvilink, 2018-07-20:
@k-okada Hi, I tried the calibrate_energy_theshold example, there is no error message so I assume it is working? 

Thanks.

closed 2018-08-09 by auvilink