jsk-ros-pkg/jsk_3rdparty #147: ros_speech_recognition:  gstreamer: Could not open audio device for recording.

auvilink, 2018-07-19:
roslaunch ros_speech_recognition speech_recognition.launch

[ERROR] [1531983914.115945111]: gstreamer: Could not open audio device for recording.
[audio_capture-2] process has died [pid 3095, exit code -11, cmd /home/pi/catkin_ws/devel/lib/audio_capture/audio_capture __name:=audio_capture __log:=/home/pi/.ros/log/aa2721dc-8b1f-11e8-9dc0-b827eb6f8a85/audio_capture-2.log].
log file: /home/pi/.ros/log/aa2721dc-8b1f-11e8-9dc0-b827eb6f8a85/audio_capture-2*.log
[audio_capture-2] restarting process

The following program / test work fine though

**python -m speech_recognition

python example/microphone_recognition.py**

I understand this is just a warper of python SpeechRecognition package for ros, wonder what could go wrong.

I use respeaker usb 4 mic array V2.0 with raspberry pi 3B.

furushchev, 2018-07-19:
This is actually not an error from the `ros_speech_recognition` node, but the `audio_capture`.
It might be helpful to look at  https://github.com/ros-drivers/audio_common .

I have no idea about what is the actual cause of the error you have but I guess that the GStreamer could not detect your microphone device.
(The `audio_capture` uses `GStreamer` as a backend of audio interface)
I at least recommend to do as follows:
1. check if your device is correctly working as a normal microphone device without ROS.
2. check if you can use your microphone device with gstreamer command. ( i.e.: `gst-launch -e alsasrc ! queue ! qtmux ! filesink location=/tmp/foo.mp4` and play the sound `/tmp/foo.mp4`.)

If you can check everything above works without any problem on your environment, then please create an issue there, since it looks that the actual problem is from the  `audio_capture` node itself, not from the backend.

auvilink, 2018-07-20:
1. check if your device is correctly working as a normal microphone device without ROS.

This has no problem, I can use python speech recognition package and run all the examples.

2. check if you can use your microphone device with gstreamer command.

gst-launch-1.0 -e alsasrc ! queue ! qtmux ! filesink location=/tmp/foo.mp4
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: Could not open audio device for recording.
Additional debug info:
gstalsasrc.c(744): gst_alsasrc_open (): /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0:
Recording open error on device 'default': No such file or directory
Setting pipeline to NULL ...
Freeing pipeline ...

auvilink, 2018-07-20:
If I add device=plughw:1,0, then 
gst-launch-1.0 -e alsasrc device=plughw:1,0 ! queue ! qtmux ! filesink location=/tmp/foo.mp4
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstAudioSrcClock
Redistribute latency...

Does that mean it is working?

auvilink, 2018-07-20:
I changed .asoundrc file and force to use usb mic, now it works.

closed 2018-07-20 by auvilink