jsk-ros-pkg/jsk_3rdparty #148: ROS Speech Recognition with Respeaker USB 4 Mic Array

auvilink, 2018-07-20:
Hi, there,

I am using Respeaker USB 4 Mic Array (you might have the same mic array) for ROS speech recognition.

This mic array has 6 channel audio output, so which channel does this ROS speech recognition actually use?  

As far as I understand, the first channel (channel 0) is the one processed for asr, any chance we can use that channel?

Thank you.

furushchev, 2018-07-21:
No configuration is needed since the first channel (channel 0) is used by default.

auvilink, 2018-07-21:
Great. Thanks.

closed 2018-07-21 by auvilink

auvilink, 2018-08-08:
Dear Mr. Furuta,

Thank you for proving such a nice wrapper for ros speech recognition.

However, some recognition result is not as expected, maybe there are some parameters I should tune?

The wave file are attached, both case, google just recognized them as: "ok, google", the latter half speech simply get ignored.

Any suggestions? Thank you so much.

Michael