jsk-ros-pkg/jsk_3rdparty #149: ros_speech_recognition Error: TypeError: listen() got an unexpected keyword argument 'phrase_time_limit

auvilink, 2018-07-21:
default installation,  roslaunch ros_speech_recognition speech_recognition.launch

line 255, in speech_recognition_srv_cb\n    src, timeout=self.listen_timeout, phrase_time_limit=self.phrase_time_limit)\n', "TypeError: listen() got an unexpected keyword argument 'phrase_time_limit'\n"]

Speech Recognition Python is installed 

https://github.com/respeaker/get_started_with_respeaker/wiki/Use-speech_recognition-python-library

pip install https://github.com/xiongyihui/speech_recognition/archive/respeaker-lite.zip

python -m speech_recognition works fine.

Any ideas?

Thank you so much.

furushchev, 2018-07-21:
@auvilink It looks you got the wrong python module with the same name installed.
We assume https://github.com/Uberi/speech_recognition for the backend which can be installed via `pip install SpeechRecognition`.
Please uninstall the existing one first, and then reinstall the correct one.

auvilink, 2018-07-21:
Thank you so much @furushchev . The one provided by respeaker is version 3.4.0, which contains the bug (phrase_time_limit) . Installed the latest official version 3.8.1 and now speech recognition works!!! :-)