jsk-ros-pkg/jsk_3rdparty #150: ros_speech_recognition only sends part of the speech for recognition

auvilink, 2018-08-08:
Hi, Mr. Furuta

I use ros_speech_recognition for speech recognition with all default settings.

I used "**continuous**" mode

[image]

However, I noticed a lot of times, it only records part of the speech and sends out for recognition, for example, if I say "ok, coco, turn left", it sends out silence + "ok, coco, turn", the last part " left" does not get send out to google for recognition.

Any parameters I should change in "SpeechRecognition.cfg"? 

Thank you so much.

furushchev, 2018-08-12:
Sorry for late reply.

> it only records part of the speech and sends out for recognition

Yes, it is due to the limitation of APIs.

> Any parameters I should change in "SpeechRecognition.cfg"?

We don't use the speech recognition on "continuous" mode so much in our use case, but yes you can still tune parameters for better results.
Please adjust parameters for your environment, for example, by using `rqt_reconfigure`.
The parameters I recommend to tune are `pause_threshold` and `non_speaking_duration`.

auvilink, 2018-08-12:
Thanks for the reply.

I am trying to use ros_speech_recognition to do voice command on robot, (move it around using speech, like "move left", "move forward", etc). Is "**continuous**" mode the best way to achieve this? If not would you please kindly give me some suggestions?

Thanks.

auvilink, 2018-08-12:
Will try those instead. Appreciated.  :-)