jsk-ros-pkg/jsk_3rdparty #156: Release request: collada_urdf_jsk_patch on kinetic

mttamtam, 2018-11-22:
I want to use collada_urdf_jsk_patch in kinetic.
The travis of kinetic version seems to be passing, and I found that lunar version of this package has been already released.

Thanks in advance.

k-okada, 2018-11-25:
collada_urdfを試してみてください。
jsk_patchの内容は最新のassimpへのリンクだとおもいますが、それは3.1になっていて、
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/3rdparty/assimp_devel/Makefile#L5

16.04だと3.2になっています。3.2でも使えない機能があれば、それはなにかを教えてください
https://packages.ubuntu.com/search?suite=all&arch=any&searchon=names&keywords=Assimp
assimp のバージョン
14.04 3.0
16.04 3.2
18.04 4.1

また、これ以外にjsk_patchで独自にcolalda_urdfへのクリティカルなパッチがあれば、それをcollada_urdfへPRを送っておいてください。

最終的にリリースが必要でも、これこれの機能がassimp,
collada_urdfへ反映されるまでの措置、という風にjsk_patchが必要な理由を明確にしてリリースしたいです

Cc: @Yohei Kakiuchi <youhei@jsk.imi.i.u-tokyo.ac.jp>
--
◉ Kei Okada

mttamtam, 2018-12-08:
すいません、collada_urdf_jsk_patchが推奨されていると思っていました。
kineticで普通にcollada_urdfで問題なく変換できましたので必要なかったです。ごめんなさい

closed 2018-12-08 by mttamtam