jsk-ros-pkg/jsk_3rdparty #157: ros_speech_recognition with GoogleCloud problem with credential_json

jonasius, 2019-01-29:
There is something messed up. Can't authenticate with '~google_cloud_credentials_json' param set to path of credential json file.

iory, 2019-01-31:
I confirmed this error. If we use google cloud api in speech recognition node, following error has occured.
```
[ERROR] [1548910020.275349]: Error processing request: recognize_google_cloud() got an unexpected keyword argument 'credential_json'
['Traceback (most recent call last):\n', '  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 625, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/home/iory/catkin_ws/jsk_3rdparty/src/jsk_3rdparty/ros_speech_recognition/scripts/speech_recognition_node.py", line 264, in speech_recognition_srv_cb\n    result = self.recognize(audio)\n', '  File "/home/iory/catkin_ws/jsk_3rdparty/src/jsk_3rdparty/ros_speech_recognition/scripts/speech_recognition_node.py", line 205, in recognize\n    return recog_func(audio_data=audio, language=self.language, **self.args)\n', "TypeError: recognize_google_cloud() got an unexpected keyword argument 'credential_json'\n"]
```

knorth55, 2020-09-08:
this is solved in #158

closed 2020-09-08 by knorth55