jsk-ros-pkg/jsk_3rdparty #174: Please add website to ff package.xml

jwhendy, 2019-09-03:
Greetings,

I'm working on automating build file generation for ROS packages on arch linux. I noticed that your website field was empty via some ROS data sources, and they indicated this comes from your [package.xml](https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/c952692139fc6690854ab8257f8cf33c5b9fdd0b/3rdparty/ff/package.xml#L10). Could you update that with a website you'd like people to come to for `ff`? For most of your other packages, that appears to be the ROS wiki.

Thanks!

k-okada, 2019-09-04:
Update url section to ff/package.xml and others ->
https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/175
You're also welcome to create PR for this.
--
◉ Kei Okada

jwhendy, 2019-09-04:
I'd be happy to PR in general. For this case, it seemed to take more time to ask you what you'd like in the file, make the change, PR, and have you merge it vs making the change directly.

PR #175 (2019-11-21): add website to package.xml
- remove whitespace in <name> section of nlopt/package.xml
- update url/website in package.xml, closes #174
commit 25376b1e (2019-11-21): update url/website in package.xml, closes #174
closed 2019-11-21 by k-okada