jsk-ros-pkg/jsk_3rdparty #187: How to use audio raw data?

Kijung-Kim, 2020-03-26:
How to use audio raw data?

I can not find pattern.

I am waiting for any advice.

thanks for reading.

knorth55, 2020-03-26:
which package are you talking about?
respeaker_ros? ros_speech_recognition?
if you can paste your code, it would be helpful for us to make good advice.

Kijung-Kim, 2020-03-26:
"respeaker_ros" is publish 
topic name is "audio" and "speech_audio"

thanks for  reply

Kijung-Kim, 2020-03-26:
> which package are you talking about?
> respeaker_ros? ros_speech_recognition?
> if you can paste your code, it would be helpful for us to make good advice.

"respeaker_ros"
topic name is "audio" and "speech_audio"

thanks for reply

708yamaguchi, 2020-03-26:
`respeaker_node` uses 16 bit audio data, but `/audio and /speech_audio (audio_common_msgs/AudioData)` uses uint8 audio data.
Therefore, you need to convert the data type uint8 <-> 16 bit.
respeaker_node:
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/respeaker_ros/scripts/respeaker_node.py#L226
audio_common_msgs/AudioData:
http://docs.ros.org/jade/api/audio_common_msgs/html/msg/AudioData.html

For the convert, you can refer to these functions:
16 bit -> uint8 (string):
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/respeaker_ros/scripts/respeaker_node.py#L282
uint8 -> 16 bit:
https://github.com/jsk-ros-pkg/jsk_recognition/pull/2478/files#diff-050e42c1f34e0ebf9955769ce6cc4245R65-R66

Kijung-Kim, 2020-03-27:
> `respeaker_node` uses int16 audio data, but `/audio` and `/speech_audio (audio_common_msgs/AudioData)` uses uint8 audio data.
> Therefore, you need to convert the data type (uint8 <-> int16).
> int16, respeaker_node:
> https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/respeaker_ros/scripts/respeaker_node.py#L254
> uint8, audio_common_msgs/AudioData:
> http://docs.ros.org/jade/api/audio_common_msgs/html/msg/AudioData.html
> 
> For the convert, you can refer to these functions:
> int16 -> uint8 (string):
> https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/respeaker_ros/scripts/respeaker_node.py#L282
> uint8 -> int16:
> https://github.com/jsk-ros-pkg/jsk_recognition/pull/2478/files#diff-050e42c1f34e0ebf9955769ce6cc4245R65-R66

Thank you for your kind answer.

knorth55, 2020-03-27:
@Kijung-Kim did you solve your problem? if so, please close this issue.
If you have another problem, feel free to open another one.

closed 2020-03-27 by Kijung-Kim