jsk-ros-pkg/jsk_3rdparty #190: sesame_ros and jsk_3rdparty regression on kinetic 32bit

tfoote, 2020-04-15:
sesame_ros has started to fail to build on kinetic 32bit release platforms. 

http://build.ros.org/view/Kbin_uX32/job/Kbin_uX32__sesame_ros__ubuntu_xenial_i386__binary/

The first failure appears to be correlated to the release of catkin_virtualenv 0.5.2 https://github.com/ros/rosdistro/pull/24374 @paulbovbel

knorth55, 2020-04-16:
```
23:19:20 ERROR: Exception:
23:19:20 Traceback (most recent call last):
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/cli/base_command.py", line 153, in _main
23:19:20     status = self.run(options, args)
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/commands/install.py", line 382, in run
23:19:20     resolver.resolve(requirement_set)
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/legacy_resolve.py", line 201, in resolve
23:19:20     self._resolve_one(requirement_set, req)
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/legacy_resolve.py", line 365, in _resolve_one
23:19:20     abstract_dist = self._get_abstract_dist_for(req_to_install)
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/legacy_resolve.py", line 313, in _get_abstract_dist_for
23:19:20     req, self.session, self.finder, self.require_hashes
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/operations/prepare.py", line 224, in prepare_linked_requirement
23:19:20     req, self.req_tracker, finder, self.build_isolation,
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/operations/prepare.py", line 49, in _get_prepared_distribution
23:19:20     abstract_dist.prepare_distribution_metadata(finder, build_isolation)
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/distributions/source/legacy.py", line 37, in prepare_distribution_metadata
23:19:20     self._setup_isolation(finder)
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_internal/distributions/source/legacy.py", line 90, in _setup_isolation
23:19:20     reqs = backend.get_requires_for_build_wheel()
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_vendor/pep517/wrappers.py", line 152, in get_requires_for_build_wheel
23:19:20     'config_settings': config_settings
23:19:20   File "/tmp/binarydeb/ros-kinetic-sesame-ros-2.1.15/obj-i686-linux-gnu/venv/local/lib/python2.7/site-packages/pip/_vendor/pep517/wrappers.py", line 255, in _call_hook
23:19:20     raise BackendUnavailable(data.get('traceback', ''))
23:19:20 BackendUnavailable
```

k-okada, 2020-04-16:
@knorth55 I'm working on this. you can concentrate on your thesis.

commit 13fd71f9 (2020-04-16): specify python module number, closes #190
closed 2020-04-16 by k-okada