jsk-ros-pkg/jsk_3rdparty #199: Failed to process package 'collada_urdf_jsk_patch'

EchChou, 2020-05-03:
I am compiling the jsk_3rdparty with catkin_make_isolated , I got this error :
**[100%] Generating /home/yusuf/catkin_ws/devel_isolated/collada_urdf_jsk_patch/lib/collada_urdf_jsk_patch/urdf_to_collada
make[3]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
HEAD is now at bd4fc36 1.11.13
patching file collada_urdf/CMakeLists.txt
patching file collada_urdf/src/collada_to_urdf.cpp
patching file collada_urdf/src/collada_urdf.cpp
patched
/bin/sh: 1: catkin: not found
/bin/sh: 1: catkin: not found
/home/yusuf/catkin_ws/src/jsk_3rdparty/jsk_ros_patch/collada_urdf_jsk_patch/Makefile:16: recipe for target 'urdf_to_collada' failed
make[3]: *** [urdf_to_collada] Error 127
CMakeFiles/urdf_to_collada.dir/build.make:60: recipe for target '/home/yusuf/catkin_ws/devel_isolated/collada_urdf_jsk_patch/lib/collada_urdf_jsk_patch/urdf_to_collada' failed
make[2]: *** [/home/yusuf/catkin_ws/devel_isolated/collada_urdf_jsk_patch/lib/collada_urdf_jsk_patch/urdf_to_collada] Error 2
CMakeFiles/Makefile2:131: recipe for target 'CMakeFiles/urdf_to_collada.dir/all' failed
make[1]: *** [CMakeFiles/urdf_to_collada.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'collada_urdf_jsk_patch': 
  Command '['/home/yusuf/catkin_ws/devel_isolated/cleaning_robot/env.sh', 'make', '-j2', '-l2']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/yusuf/catkin_ws/build_isolated/collada_urdf_jsk_patch && /home/yusuf/catkin_ws/devel_isolated/cleaning_robot/env.sh make -j2 -l2

Command failed, exiting.**
Someone can help please ?
Thank you.

ishiguroJSK, 2020-05-03:
I don't know well about this repository, but I can see  `/bin/sh: 1: catkin: not found`.
I guess you haven't installed catkin related commands.
How about installing `sudo apt install ros-melodic-catkin python-wstools python-catkin-tools`

pazeshun, 2020-05-04:
@ishiguroJSK ’s comment is partially wrong.
You only need `python-catkin-tools`.
And this dependency should be automatically resolved by `rosdep install` command.
Please use `rosdep install` before building any ROS package.
http://wiki.ros.org/rosdep#Install_dependency_of_all_packages_in_the_workspace
(If you want to resolve dependency of only one package, you have to specify the path of the package after `--from-paths` instead of `src`)

closed 2025-06-13 by mqcmd196