jsk-ros-pkg/jsk_3rdparty #201: Error processing request: global name 'sound' is not defined

MrMarshy, 2020-06-08:
ros_speech_recoginition_node.py

Line: 178

 if hasattr(SoundRequest, 'volume'): # volume is added from 0.3.0 https://github.com/ros-drivers/audio_common/commit/da9623414f381642e52f59701c09928c72a54be7#diff-fe2d85580f1ccfed4e23a608df44a7f7
            sound.volume = 1.0

I think this should be req.volume. Throws error:

Error processing request: global name 'sound' is not defined
['Traceback (most recent call last):\n', '  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 632, in _handle_request\n    response = convert_return_to_response(self.handler(request), self.response_class)\n', '  File "/opt/ros/melodic/share/ros_speech_recognition/scripts/speech_recognition_node.py", line 257, in speech_recognition_srv_cb\n    self.play_sound("start", 0.1)\n', '  File "/opt/ros/melodic/share/ros_speech_recognition/scripts/speech_recognition_node.py", line 179, in play_sound\n    sound.volume = 1.0\n', "NameError: global name 'sound' is not defined\n"]

708yamaguchi, 2020-06-08:
Thank you for your report.

> sound.volume = 1.0
> 
> I think this should be req.volume.

I think your point is correct.

Could you please create a pull request to modify it?
If it's not possible, I'll do it.

knorth55, 2020-09-08:
this is solved in #202

closed 2020-09-08 by knorth55