jsk-ros-pkg/jsk_3rdparty #219: nlopt still required?

mgruhler, 2020-09-08:
Starting with ROS noetic / Ubuntu focal, the system dependency `libnlopt-cxx-dev` provides access to the latest version on `nlopt` (currently 2.6.1), as well as proper cmake files to easily include them.

A [PR](https://github.com/ros/rosdistro/pull/26446) to add a `rosdep` key has been opened too.

We previously used your package but, in focal, stumbled upon segfaults in nlopt that we couldn't track down and thought that maybe it is an issue with being out of date.

I suggest, if you aren't specifically requiring the included 2.3.x version, to stop distributing this package and switch to the system dependency.

k-okada, 2020-09-09:
Thank you. I have released newer version without nlopt
https://github.com/ros/rosdistro/pull/26499

closed 2020-09-09 by k-okada

mgruhler, 2020-09-09:
@k-okada Thanks. That was quick :-)
Hope this isn't breaking anything on your end.

Note that there is currently no rosdep key, as the linked PR is not yet merged...