jsk-ros-pkg/jsk_3rdparty #228: aques_talk cannot speak number and alphabets

knorth55, 2020-12-19:
aques_talk cannot speak numbers now.
we can use tag value for numbers and alphabets.
we need to replace them in `text2wave`.

```bash
<NUM VAL=10>
<NUMK VAL=10 COUNTER=ふん>
<ALPHA VAL=fetch>
```

cc. @mqcmd196 @708yamaguchi 
https://www.a-quest.com/archive/manual/siyo_onseikigou.pdf

708yamaguchi, 2020-12-19:
I'm sorry, but I can't open the PDF.
Chrome is saying "Error Failed to load PDF document".
I'll try again after some time.

708yamaguchi, 2020-12-19:
I can open the PDF now.
I place the file at Manuals -> JSK robot system in our Google Drive.

708yamaguchi, 2020-12-19:
I created PR https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/229

708yamaguchi, 2020-12-21:
I checked that fetch1075 can speak number by
```bash
$ rosrun jsk_fetch_startup time_signal.py
```

However, fetch1075 speaks "20ときです"　while the input is "20時です"
This is because I do not consider 助数詞.
I will try to solve this problem.

closed 2021-06-15 by knorth55