jsk-ros-pkg/jsk_3rdparty #250: [google_cloud_texttospeech] ディレクトリ構成が違うと日本語を話さない

Naoki-Hiraoka, 2021-05-31:
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/2b0226ec32e76176eff179f61f7dd77461e3fa4e/3rdparty/google_cloud_texttospeech/launch/google_cloud_texttospeech.launch#L42

Naoki-Hiraoka, 2021-05-31:
$(find google_cloud_texttospeech)は，launchを呼んだPCでの`google_cloud_texttospeech`へのパスが入るため，sound_play_machineに指定したPCとディレクトリ構成が違うと正しくパスを指定できませんでした．

knorth55, 2021-05-31:
この場合はmachine tagのenv-loaderに書く必要があるかと思います．

knorth55, 2021-05-31:
いや違うか．
env-loaderは更新する必要はないのかな．
remoteマシーンでjsk_3rdpartyがインストールされているか，最新版になっているかを確認してください．
あとIssueをたてるときはlog outputを貼ってください．

Naoki-Hiraoka, 2021-05-31:
roslaunchしたPC側の
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/2b0226ec32e76176eff179f61f7dd77461e3fa4e/3rdparty/google_cloud_texttospeech/launch/google_cloud_texttospeech.launch#L42
の行を編集して，sound_play_machineに指定したPCの`google_cloud_texttospeech`へのパスをベタ書きすると動くことは確認できました．

env-loaderでPATHを指定しても，roslaunchのenvタグのPATHによって上書きされるため，効果はありませんでした．

```
SIOD ERROR: unbound variable : ja
[ERROR] [1622449077.811238]: Sound synthesis failed. Is festival installed? Is a festival voice installed? Try running "rosdep satisfy sound_play|sh". Refer to http://wiki.ros.org/sound_play/Troubleshooting
[ERROR] [1622449077.814098]: Exception in actionlib callback: 'NoneType' object has no attribute 'command'
```

machineタグを使用する場合に，roslaunchの`$(find )`をremote PC側で評価してくれればよいのですが...

machineタグを使うことは諦めて,remote PCにsshしてからroslaunchすることにします...

knorth55, 2021-05-31:
なんとなく言っているエラーはわかりましたが，logを全て貼ってください．話はそこから始まるので．
そして，`roslaunch`の`find`がlocal側で評価されているということですね．
それじゃあbashスクリプトをmachineタグで実行するのはどうでしょうか？
https://github.com/jsk-ros-pkg/jsk_robot/blob/master/jsk_pr2_robot/jsk_pr2_startup/jsk_pr2_sensors/kinect_head.launch#L49-L60

knorth55, 2021-06-04:
@Naoki-Hiraoka この問題どうなりましたか？

Naoki-Hiraoka, 2021-06-04:
自分個人の事情としては、今回自分が`machine`タグを使いたかった理由は、これまでremote PCにsshしてから`roslaunch`する独自bashスクリプトを書いていたのですが、この独自bashスクリプトを使うのをやめたかったからでした。しかし、`roslaunch`の`find`がlocal側で評価されてしまうため`machine`タグを使用したとしても結局独自bashスクリプトは必要と分かったため、`machine`タグを使うのをやめて現状維持としました。

`google_cloud_texttospeech`のlaunchファイルの引数`sound_play_machine`がうまく機能しないというこのissueの問題は依然として残っています。

knorth55, 2021-06-05:
現状維持というのは，わざわざsshしてlaunchする方法でしょうか？
それならmachinタグでbashスクリプトするほうが工程が減っていい気がしますが

Naoki-Hiraoka, 2021-06-05:
説明が曖昧でした。「remote PCにsshしてからroslaunchする」という処理を行うbashスクリプトです。

knorth55, 2021-06-05:
それだとしても，ターミナルで起動するコマンドが１つになるのでは？
それともsshしてroslaunchするスクリプトをノードとして実行しているってこと？
なら，マシンタグにすればいいのでは，とおもうんだけど．

Naoki-Hiraoka, 2021-06-05:
現物を直接見て頂いた方が早そうです。privateなリポジトリにあります。
https://github.com/start-jsk/rtmros_hrp2/issues/591

knorth55, 2021-06-05:
うん，だからmachineタグでこの他のlaunchに組み込んだら良くないかな．
起動するlaunchを減らしたいという話ではないの？

Naoki-Hiraoka, 2021-06-05:
https://github.com/start-jsk/rtmros_hrp2/issues/591#issuecomment-855225253

Naoki-Hiraoka, 2021-06-05:
`google_cloud_texttospeech`のlaunchファイルの引数`sound_play_machine{がうまく機能しないというこのissueの対象のバグと論点が外れてきましたので、https://github.com/start-jsk/rtmros_hrp2/issues/591 に移動しませんか。

knorth55, 2021-06-05:
なるほど、もとから１つのlaunchなのね。
じゃあそれでいいんじゃないかな。
問題としてはlaunchの`find`が遅延評価されていないってことかな。