jsk-ros-pkg/jsk_3rdparty #255: 「！」や「『』」のような記号があると、sound synthesis failedと出てしまい、応答を読み上げてくれな?

k-okada, 2021-06-03:
? また、応答に「！」や「『』」のような記号があると、sound synthesis failedと出てしまい、応答を読み上げてくれないことがある印象だったので、気がついたものに関しては置換しています。

は，aques_talk の問題？voice_text だと起こらない？など条件を知りたいのと，
もしそうだたすると，aques_talk 側のプログラムで直したいです (https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/a5c20df492cf488d218ce3a81db31691cc720a8a/chaplus_ros/scripts/chaplus_ros.py#L138-L141)

cc: @ayfujii

Naoki-Hiraoka, 2021-06-03:
https://github.com/jsk-ros-pkg/jsk_3rdparty/tree/master/3rdparty/google_cloud_texttospeech は問題なく応答を読み上げてくれることを確認しました。
```
hiraoka@hiraoka-desktop:/opt/ros/melodic/share/pr2eus$ roseus ./speak.l 
$ (ros::roseus "aaa") 
$ (speak-jp "『こんにちは』") 
$ (speak-jp "こんにちは！") 
$ (speak-jp "こんにちは〜") 
```

708yamaguchi, 2021-06-03:
この件に関して、aques_talkにプルリクエストを出しました。
https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/257

まずaques_talkが「！」や「こんにちは！」を読めない問題ですが、これは特殊なタグで囲うことで読めるようになります。
https://www.a-quest.com/archive/manual/siyo_onseikigou.pdf
この音声記号列仕様表とAquesTalk2 Linux Manualの存在をREADMEに書き、かつ、JSKのGoogle Driveにもおいておきました。
https://www.a-quest.com/archive/manual/aqtk2_lnx_man.pdf

このように特定の文字列を特殊タグで囲う作業はすでにaques_talkでは行われていて、以下のように入力文字列をsedで置き換えることで、数字やアフファベットを読めるようにしています。
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/2b0226ec32e76176eff179f61f7dd77461e3fa4e/3rdparty/aques_talk/text2wave#L33-L41

今回は「！」を読んでほしいわけではないと思うので、sedの行を追加して「！」や「『」などの文字を無視するようにしました。
ただ、「'」や「"」が文章の開始文字として認識されたり、「[」や「]」がsedの制御文字と認識されたりと、無視しきれない文字があったので、README.mdにそのことを書いておきました。（SQLインジェクション対策の重要性がわかってきました）
僕の正規表現力が低い可能性が高いので、実は「]」は入力できるよ、のような表現があれば教えていただきたいです。

次にvoice_textで(send *ri* :speak-jp xxx)を行ったところ、「！」は読めなかったですが「こんにちは！」を読むことは出来ました。また、fetch_bringup.launchのログにエラーも表示されませんでした。ReakSpeaker（旧voice_text）のオンライン音声合成サンプルでも試しましたが、やはり「！」は読めず「こんにちは！」を読むことは出来ました。https://readspeaker.jp/

僕は今のところ、voice_textではAquesTalkのように音声記号列仕様表のようなドキュメントを見つけられていないので、見つけ次第追加で返信します。

ayfujii, 2021-06-03:
プルリクありがとうございます。

プルリクのブランチのaques_talkを使って、上記のchaplus_ros.pyの138-141行目のreplaceしているところは消したコードで
roslaunch chaplus_ros google_example.launch
を試したところ、エラーが出なくなっていることを確認しました。

上記のchaplus_ros.pyの138-141行目は消去したいと思います。

708yamaguchi, 2021-06-03:
ご確認ありがとうございます。
その後、プログラムの見通しを良くするために追加コミットをしたので、再度ご確認していただけると助かります。

ちなみに、aques_talk側での文字列置換では以下の操作を行っています。
1. 漢字をひらがなに変換。
```
nkf -j %s | kakasi -JH | nkf -w | \
```
2. 「，」を「、」に、「,」を「、」に、「．」を「。」に、「.」を「。」に変換。
```
sed -e 's/，/、/g' | sed -e 's/,/、/g' | \
sed -e 's/．/。/g' | sed -e 's/\./。/g' | \
```
3. 正規表現で[a-zA-Z0-9ぁ-んァ-ンー、。?？]を表す文字以外を全て削除。つまり、アルファベット、数字、ひらがな、カタカナ、「ー、。?？」以外を全て削除。
```
sed -e 's/[^a-zA-Z0-9ぁ-んァ-ンー、。?？]//g' | \
```
4. AquesTalk2がアルファベットや数字を発音できるように、特殊な制御文字を追加。
```
sed -e 's/\([a-zA-Z]\+\)/<ALPHA VAL=\\1>/g' | \
sed -e 's/\([0-9]\+\)/<NUMK VAL=\\1>/g' > \
```

ayfujii, 2021-06-03:
ありがとうございます。動作することを確認しました。

細かい話になってしまうのですが、「！」は文の切れ目に多いと思うので「。」に置換するのと、「〜」は伸びる音「ー」に置換したほうが、よりもともとのニュアンスが伝わるかなと思いました。
下記のものを追加するのはいかがでしょうか？
ご検討よろしくお願い致します。
```
sed -e 's/！/。/g' | sed -e 's/\!/。/g' | \
sed -e 's/〜/ー/g' | \ 
```

k-okada, 2021-06-03:
@ayfujii よい提案だと思うので
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/57c53ad08b09709b88538c3319153c08aff65cea/3rdparty/aques_talk/README.md
にpull requestを送るのが良いと思います

その際、それぞれの場合の音声のwavファイルもつけると説得力が増します