jsk-ros-pkg/jsk_3rdparty #265: Separate ros_speech_recognition repo

roboav8r, 2021-06-29:
Hi all,
I am using ```ros_speech_recognition``` for an offline speech recognition project. However, I'm not using the other packages in the ```jsk_3rdparty``` repo.

Is there a separate repo which just contains ```ros_speech_recognition```? If not, would it be possible to make a separate ```ros_speech_recognition``` repo that is not part of ```jsk_3rdparty``` or set ```ros_speech_recognition``` up as a submodule of ```jsk_3rdparty```? 

Thanks!

knorth55, 2021-06-29:
> Is there a separate repo which just contains ros_speech_recognition? If not, would it be possible to make a separate ros_speech_recognition repo that is not part of jsk_3rdparty or set ros_speech_recognition up as a submodule of jsk_3rdparty?

No.
we cannot separate the repo because it make maintenance difficult.
you can use `ros_speech_recognition` from debian package `ros-melodic-ros-speech-recognition` or you can clone the directory only with `git>2.9`.
https://stackoverflow.com/questions/600079/how-do-i-clone-a-subdirectory-only-of-a-git-repository/52269934#52269934

roboav8r, 2021-06-29:
Thank you for the information, I know it can be tough to manage large repos. I have the most recent version of Git so I will clone just the subdirectory.

closed 2021-06-29 by roboav8r

knorth55, 2021-06-29:
sorry I made a typo. you need git>=2.19` for subdirectory cloning.
another option is putting `CATKIN_IGNORE` to unnecessary packages to avoid build failure.