jsk-ros-pkg/jsk_3rdparty #268: [google_cloud_texttospeech] Cannot change speaking_rate.

iory, 2021-07-12:
`text2wave` reads `speaking_rate` at first and sample launch file gives `speaking_rate` 
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/3rdparty/google_cloud_texttospeech/launch/google_cloud_texttospeech.launch#L46 .
However, it seems that this `speaking_rate` cannot be given correctly.

knorth55, 2021-07-12:
>However, it seems that this speaking_rate cannot be given correctly.

this line does not have the effect?
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/27767817252febb8322a291be36042bc5c51c76f/3rdparty/google_cloud_texttospeech/bin/text2wave#L10

iory, 2021-07-12:
`text2wave` is not a rosnode. It is called from `os.system` in `sound play node`.
Unfortunately, the ros namespace is not carried over to `text2wave`.

iory, 2021-07-12:
The output for these change (https://github.com/iory/jsk_3rdparty/blob/268/3rdparty/google_cloud_texttospeech/bin/text2wave#L11-L12 ) is as follows.
```
[INFO] [1626085226.732977]: sound_play node is ready to play sound
[Text2Wave] Speaking rate is 1.0
[Text2Wave] namespace "/"
```

knorth55, 2021-09-17:
@knorth55  in order to solve this issue, we need to make the tts module as plugin in `sound_play`.