jsk-ros-pkg/jsk_3rdparty #356: Request for next release 2.1.25

knorth55, 2022-07-14:
`switchbot_ros` fails building on `build.ros.org` in `2.1.24`.

https://build.ros.org/job/Mbin_uB64__switchbot_ros__ubuntu_bionic_amd64__binary/
https://build.ros.org/job/Mbin_ubhf_uBhf__switchbot_ros__ubuntu_bionic_armhf__binary/
https://build.ros.org/job/Mbin_ubv8_uBv8__switchbot_ros__ubuntu_bionic_arm64__binary/

this build error may be caused by missing `message_generation` dependencies in `package.xml`.
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/622d2c03b5421fb9de83c60c080cc11fcca0ad77/switchbot_ros/package.xml#L12-L17

this missing dependency problem is solve in https://github.com/jsk-ros-pkg/jsk_3rdparty/commit/d7c52ccd601c930da8e9e2cdd38fcfd893234377, so the package build may be succeeded in next release.

cc. @mqcmd196 @sktometometo

knorth55, 2022-07-14:
I didn't check deeply about the build error, so we need to check the build error deeply.

knorth55, 2022-07-14:
I checked a little bit and I add more missing dependencies in #357 

related: #357 #358

mqcmd196, 2022-07-15:
Thank you for your check, I confirmed that I can get the email, so I'll check the buildfarm carefully.

knorth55, 2022-07-15:
Now I can reproduce the same error of `build.ros.org` locally.

```bash
docker run -it osrf/ros:melodic-desktop-bionic /bin/bash
# in docker
apt update
apt install python-bloom
apt install fakeroot dpkg-dev debhelper

cd 
git clone https://github.com/jsk-ros-pkg/jsk_3rdparty.git
cd jsk_3rdparty/switchbot_ros
bloom-generate rosdebian --os-name ubuntu --os-version bionic --ros-distro melodic
apt purge ros-melodic-std-msgs ros-melodic-actionlib-msgs

dpkg-buildpackage -rfakeroot -uc -b

```

knorth55, 2022-07-15:
I carefully read the build log and I found the following error in `build.ros.org`.
https://build.ros.org/job/Mbin_uB64__switchbot_ros__ubuntu_bionic_amd64__binary/364/consoleFull#console-section-8

```
15:04:22 -- Could NOT find actionlib_msgs (missing: actionlib_msgs_DIR)
15:04:22 -- Could not find the required component 'actionlib_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
15:04:22 CMake Error at /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
15:04:22   Could not find a package configuration file provided by "actionlib_msgs"
15:04:22   with any of the following names:
15:04:22 
15:04:22     actionlib_msgsConfig.cmake
15:04:22     actionlib_msgs-config.cmake
15:04:22 
15:04:22   Add the installation prefix of "actionlib_msgs" to CMAKE_PREFIX_PATH or set
15:04:22   "actionlib_msgs_DIR" to a directory containing one of the above files.  If
15:04:22   "actionlib_msgs" provides a separate development package or SDK, be sure it
15:04:22   has been installed.
15:04:22 Call Stack (most recent call first):
15:04:22   CMakeLists.txt:4 (find_package)
```