jsk-ros-pkg/jsk_3rdparty #359: support mebo?

k-okada, 2022-07-19:
https://qiita.com/maKunugi/items/14f1b82a2c0b6fa5c202

ChatPlusが8/31 で終了みたいです。
のchatplus_ros は　A3RT もサポートしているみたいだけど，普段どっち使ているのかな？

デフォルトを変えておいた方が良いのと，必要だったら meboも対応させるのが良いですね。

https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/07de833b4810dfd0161fcbab7bda6a8d25c471b2/chaplus_ros/sample/google_example.launch#L4
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/07de833b4810dfd0161fcbab7bda6a8d25c471b2/chaplus_ros/scripts/chaplus_ros.py#L55
https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/chaplus_ros/README.md#parameters

MiyabiTane, 2022-07-19:
私はChaplusを使っていたので、近々meboも対応させようと思います。
昨日、エージェントを作成してAPI Keyを発行するところまで行いましたが、chaplusよりも自然な会話ができているように思えました。

対応したらPRさせて頂きます。よろしくお願い致します。

MiyabiTane, 2022-07-19:
@ayfujii 
[README](https://github.com/jsk-ros-pkg/jsk_3rdparty/tree/master/chaplus_ros#a3rt)に載っているA3RT関連のリンクが見れなかったのですが、これらはまだサポートされているのでしょうか？？ご確認頂ければ幸いです。よろしくお願い致します。

ayfujii, 2022-07-19:
ありがとうございます。
A3RTはリンクが切れているのに気がつきませんでした。
ドメインが変更となっているようですが、API KEYなどはそのままで使えました。
[こちら](https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/360)にプルリクエストを送りました。

MiyabiTane, 2022-07-19:
早速のご対応ありがとうございます！！
リンクを見ることができました、まだサポートされているようでよかったです。meboがあまりよくなさそうだったらこちらも試してみようと思います。

MiyabiTane, 2022-08-16:
大変遅くなりましたすみません。meboに対応しました。
https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/374

会話を重ねるごとに賢くなっていく仕様のようなので、私が作成したエージェントを共有で使用するのが良いかと思い、ドライブにAPI Keyを置きました。
1つのAPIを皆で使うという仕様で良いかということも含め、レビューして頂けると嬉しいです。
よろしくお願い致します