jsk-ros-pkg/jsk_3rdparty #391: [google-chat-ros] problems of google-auth

a-ichikura, 2022-11-01:
I tried to chat with Strelka(Spot) by google chat and refer this [README.md](https://github.com/mqcmd196/jsk_3rdparty/tree/google_chat_ros/google_chat_ros)

It seems I was able to create google service account, get API key and set API settings. Chapter [1.2.4](https://github.com/mqcmd196/jsk_3rdparty/tree/google_chat_ros/google_chat_ros#4-set-google-chat-api-connection-settings)

However, in chapter 1.3, catkin build failed due to the mismatch of google-auth. 

```
____________________________________________________________________________________________________
Errors     << google_chat_ros:make /home/ichikura/spot_ws/logs/google_chat_ros/build.make.000.log   
Could not find a version that matches google-auth<2.0dev,<3.0.0dev,>=1.19.0,>=1.25.0,>=2.13.0 (from -r requirements.in (line 7))
Tried: 0.0.1, 0.0.1, 0.1.0, 0.1.0, 0.2.0, 0.2.0, 0.3.0, 0.3.0, 0.3.1, 0.3.1, 0.3.2, 0.3.2, 0.4.0, 0.4.0, 0.5.0, 0.5.0, 0.6.0, 0.6.0, 0.7.0, 0.7.0, 0.8.0, 0.8.0, 0.9.0, 0.9.0, 0.10.0, 0.10.0, 1.0.0, 1.0.0, 1.0.1, 1.0.1, 1.0.2, 1.0.2, 1.1.0, 1.1.0, 1.1.1, 1.1.1, 1.2.0, 1.2.0, 1.2.1, 1.2.1, 1.3.0, 1.3.0, 1.4.0, 1.4.0, 1.4.1, 1.4.1, 1.4.2, 1.4.2, 1.5.0, 1.5.0, 1.5.1, 1.5.1, 1.6.0, 1.6.0, 1.6.1, 1.6.1, 1.6.2, 1.6.2, 1.6.3, 1.6.3, 1.7.0, 1.7.0, 1.7.1, 1.7.1, 1.7.2, 1.7.2, 1.8.0, 1.8.0, 1.8.1, 1.8.1, 1.8.2, 1.8.2, 1.9.0, 1.9.0, 1.10.0, 1.10.0, 1.10.1, 1.10.1, 1.10.2, 1.10.2, 1.11.0, 1.11.0, 1.11.1, 1.11.1, 1.11.2, 1.11.2, 1.11.3, 1.11.3, 1.12.0, 1.12.0, 1.13.0, 1.13.0, 1.13.1, 1.13.1, 1.14.0, 1.14.0, 1.14.1, 1.14.1, 1.14.2, 1.14.2, 1.14.3, 1.14.3, 1.15.0, 1.15.0, 1.16.0, 1.16.0, 1.16.1, 1.16.1, 1.17.0, 1.17.0, 1.17.1, 1.17.1, 1.17.2, 1.17.2, 1.18.0, 1.18.0, 1.19.0, 1.19.0, 1.19.1, 1.19.1, 1.19.2, 1.19.2, 1.20.0, 1.20.0, 1.20.1, 1.20.1, 1.21.0, 1.21.0, 1.21.1, 1.21.1, 1.21.2, 1.21.2, 1.21.3, 1.21.3, 1.22.0, 1.22.0, 1.22.1, 1.22.1, 1.23.0, 1.23.0, 1.24.0, 1.24.0, 1.25.0, 1.25.0, 1.26.0, 1.26.0, 1.26.1, 1.26.1, 1.27.0, 1.27.0, 1.27.1, 1.27.1, 1.28.0, 1.28.0, 1.28.1, 1.28.1, 1.29.0, 1.29.0, 1.30.0, 1.30.0, 1.30.1, 1.30.1, 1.30.2, 1.30.2, 1.31.0, 1.31.0, 1.32.0, 1.32.0, 1.32.1, 1.32.1, 1.33.0, 1.33.0, 1.33.1, 1.33.1, 1.34.0, 1.34.0, 1.35.0, 1.35.0, 2.0.0, 2.0.0, 2.0.1, 2.0.1, 2.0.2, 2.0.2, 2.1.0, 2.1.0, 2.2.0, 2.2.0, 2.2.1, 2.2.1, 2.3.0, 2.3.0, 2.3.1, 2.3.1, 2.3.2, 2.3.2, 2.3.3, 2.3.3, 2.4.0, 2.4.0, 2.4.1, 2.4.1, 2.5.0, 2.5.0, 2.6.0, 2.6.0, 2.6.2, 2.6.2, 2.6.3, 2.6.3, 2.6.4, 2.6.4, 2.6.5, 2.6.5, 2.6.6, 2.6.6, 2.7.0, 2.7.0, 2.8.0, 2.8.0, 2.9.0, 2.9.0, 2.9.1, 2.9.1, 2.10.0, 2.10.0, 2.11.0, 2.11.0, 2.11.1, 2.11.1, 2.12.0, 2.12.0, 2.13.0, 2.13.0, 2.14.0, 2.14.0
Skipped pre-versions: 2.0.0.dev0, 2.0.0.dev0, 2.0.0b1, 2.0.0b1
There are incompatible versions in the resolved dependencies:
  google-auth (from -r requirements.in (line 7))
  google-auth<2.0dev,>=1.25.0 (from google-api-core[grpc]==1.32.0->-r requirements.in (line 3))
  google-auth (from google-auth-httplib2==0.1.0->-r requirements.in (line 5))
  google-auth>=2.13.0 (from google-auth-oauthlib==0.7.0->-r requirements.in (line 6))
  google-auth<3.0.0dev,>=1.19.0 (from google-api-python-client==2.52.0->-r requirements.in (line 4))
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/catkin_virtualenv/venv_lock", line 51, in <module>
    extra_pip_args=[arg for arg in extra_pip_args.split(" ") if arg != ""],
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin_virtualenv/venv.py", line 143, in lock
    run_command(command, check=True)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin_virtualenv/__init__.py", line 44, in run_command
    return subprocess.run(cmd, *args, **kwargs)
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['/home/ichikura/spot_ws/build/google_chat_ros/venv/bin/pip-compile', '--no-header', 'requirements.in', '--pip-args', '-qq --retries 10 --timeout 30', '-o', '/home/ichikura/spot_ws/src/jsk-ros-pkg/jsk_3rdparty/google_chat_ros/requirements.txt']' returned non-zero exit status 2.
make[2]: *** [/home/ichikura/spot_ws/src/jsk-ros-pkg/jsk_3rdparty/google_chat_ros/requirements.txt] Error 1
make[1]: *** [CMakeFiles/google_chat_ros_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
cd /home/ichikura/spot_ws/build/google_chat_ros; catkin build --get-env google_chat_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
....................................................................................................
```

And Run Time Error.
```
~/spot_ws/src/jsk-ros-pkg/jsk_3rdparty/google_chat_ros/scripts$ roslaunch google_chat_ros google_chat.launch receiving_mode:=pubsub google_cloud_credentials_json:=/home/ichikura/Downloads/tribal-quasar-343006-cc9553cfe079.json project_id:=tribal-quasar-343006 subscription_id:=chat-sub
... logging to /home/ichikura/.ros/log/91c6d8a4-59a0-11ed-a062-287fcf9948c3/roslaunch-ichikura-ThinkPad-T495s-28306.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ichikura-ThinkPad-T495s:40843/

SUMMARY
========

PARAMETERS
 * /google_chat_ros/download_avatar: False
 * /google_chat_ros/download_data: True
 * /google_chat_ros/download_data_timeout: 10
 * /google_chat_ros/download_directory: /tmp
 * /google_chat_ros/gdrive_upload_service: /gdrive_ros/upload
 * /google_chat_ros/google_cloud_credentials_json: /home/ichikura/Do...
 * /google_chat_ros/host: localhost
 * /google_chat_ros/port: 443
 * /google_chat_ros/project_id: tribal-quasar-343006
 * /google_chat_ros/receiving_mode: pubsub
 * /google_chat_ros/ssl_certfile: default
 * /google_chat_ros/ssl_keyfile: default
 * /google_chat_ros/subscription_id: chat-sub
 * /google_chat_ros/upload_data_timeout: 20
 * /google_chat_ros_helper/debug_sound: False
 * /google_chat_ros_helper/to_dialogflow_client: False
 * /rosdistro: melodic
 * /rosversion: 1.14.13

NODES
  /
    google_chat_ros (google_chat_ros/google_chat_ros_node.py)
    google_chat_ros_helper (google_chat_ros/helper.py)

auto-starting new master
process[master]: started with pid [28316]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 91c6d8a4-59a0-11ed-a062-287fcf9948c3
process[rosout-1]: started with pid [28327]
started core service [/rosout]
process[google_chat_ros-2]: started with pid [28334]
process[google_chat_ros_helper-3]: started with pid [28335]
Traceback (most recent call last):
  File "/home/ichikura/spot_ws/src/jsk-ros-pkg/jsk_3rdparty/google_chat_ros/scripts/google_chat_ros_node.py", line 2, in <module>
    import gdown
ModuleNotFoundError: No module named 'gdown'
[google_chat_ros-2] process has died [pid 28334, exit code 1, cmd /home/ichikura/spot_ws/src/jsk-ros-pkg/jsk_3rdparty/google_chat_ros/scripts/google_chat_ros_node.py __name:=google_chat_ros __log:=/home/ichikura/.ros/log/91c6d8a4-59a0-11ed-a062-287fcf9948c3/google_chat_ros-2.log].
log file: /home/ichikura/.ros/log/91c6d8a4-59a0-11ed-a062-287fcf9948c3/google_chat_ros-2*.log
```
@mqcmd196

mqcmd196, 2022-11-01:
Sorry, could you update your branch? I fixed with 0e7d55298a92621020b33d77715ce28cfdb25778

a-ichikura, 2022-11-02:
Thank you for your updating.

I got warnings, but made it successfully.

```___________________________________________________________________________________________________________________________________________________________________________________________________________
Warnings   << google_chat_ros:make /home/ichikura/spot_ws/logs/google_chat_ros/build.make.001.log                                                                                                          
beautifulsoup4==4.11.1    # via gdown
cachetools==4.2.4         # via google-auth
certifi==2022.9.24        # via requests
charset-normalizer==2.0.12  # via requests
click==6.7                # via -r requirements.in
filelock==3.4.1           # via gdown
gdown==4.4.0              # via -r requirements.in
google-api-core[grpc]==2.8.2  # via -r requirements.in, google-api-python-client, google-cloud-pubsub
google-api-python-client==2.51.0  # via -r requirements.in
google-auth-httplib2==0.1.0  # via -r requirements.in, google-api-python-client
google-auth-oauthlib==0.5.2  # via -r requirements.in
google-auth==2.8.0        # via -r requirements.in, google-api-core, google-api-python-client, google-auth-httplib2, google-auth-oauthlib
google-cloud-pubsub==2.12.1  # via -r requirements.in
googleapis-common-protos[grpc]==1.56.2  # via -r requirements.in, google-api-core, grpc-google-iam-v1, grpcio-status
grpc-google-iam-v1==0.12.4  # via -r requirements.in, google-cloud-pubsub
grpcio-status==1.46.3     # via -r requirements.in, google-api-core, google-cloud-pubsub
grpcio==1.46.3            # via -r requirements.in, google-api-core, google-cloud-pubsub, googleapis-common-protos, grpc-google-iam-v1, grpcio-status
httplib2==0.21.0          # via -r requirements.in, google-api-python-client, google-auth-httplib2, oauth2client
idna==3.4                 # via requests
importlib-resources==5.4.0  # via tqdm
oauth2client==4.1.3       # via -r requirements.in
oauthlib==3.2.0           # via -r requirements.in, requests-oauthlib
proto-plus==1.22.1        # via google-cloud-pubsub
protobuf==3.19.4          # via -r requirements.in, google-api-core, googleapis-common-protos, grpcio-status, proto-plus
pyasn1-modules==0.2.8     # via google-auth, oauth2client
pyasn1==0.4.8             # via oauth2client, pyasn1-modules, rsa
pyparsing==3.0.9          # via httplib2
pysocks==1.7.1            # via requests
requests-oauthlib==1.3.1  # via -r requirements.in, google-auth-oauthlib
requests[socks]==2.27.1   # via -r requirements.in, gdown, google-api-core, requests-oauthlib
rsa==4.9                  # via google-auth, oauth2client
six==1.16.0               # via gdown, google-auth, google-auth-httplib2, grpcio, oauth2client
soupsieve==2.3.2.post1    # via beautifulsoup4
tqdm==4.64.1              # via gdown
uritemplate==4.1.1        # via google-api-python-client
urllib3==1.26.12          # via requests
zipp==3.6.0               # via importlib-resources
ERROR: pip-tools 5.1.2 has requirement click>=7, but you'll have click 6.7 which is incompatible.
cd /home/ichikura/spot_ws/build/google_chat_ros; catkin build --get-env google_chat_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Finished  <<< google_chat_ros 
```
And There is no Runtime Error.

a-ichikura, 2022-11-02:
I sent a message to Strelka app, and succeeded in subscribing  `/google_chat_ros/message_activity` and publishing `google_chat_ros/send/goal`

```
 $ rostopic echo /google_chat_ros/message_activity 
event_time: "2022-11-02T07:46:21.874751Z"
space: 
  name: "spaces/p3s05EAAAAE"
  display_name: ''
  room: False
  dm: True
message: 
  name: "spaces/<space name>/messages/<message id>"
  sender: 
    name: "users/<user id>"
    display_name: "Aiko Ichikura"
    avatar_url: "<avatar url>"
    avatar: []
    email: "<email>"
    bot: False
    human: True
  create_time: "2022-11-02T07:46:21.874751Z"
  text: !!python/str "\u3053\u3093\u306B\u3061\u306F"
  thread_name: "spaces/<space name>/threads/<thread name>"
  annotations: []
  argument_text: !!python/str "\u3053\u3093\u306B\u3061\u306F"
  attachments: []
user: 
  name: "users/<user id>"
  display_name: "Aiko Ichikura"
  avatar_url: "<avatar url>"
  avatar: []
  email: "<email>"
  bot: False
  human: True
---
```

```
$ rostopic pub -1 /google_chat_ros/send/goal google_charos/SendMessageActionGoal "goal:
>   text: 'Hello!'
>   space: 'spaces/<space name>'"
publishing and latching message for 3.0 seconds
```
[image]

mqcmd196, 2022-11-02:
Cool! Now I'm fixing CI, so please wait for merging into origin/master

closed 2022-11-02 by mqcmd196