jsk-ros-pkg/jsk_3rdparty #464: [respeaker_ros] デバイスを認識できません

heissereal, 2023-05-14:
自分のPCでデバイスを繋いだときはうまくできていたのですが、spotにrespeakerを繋いだときに、以下のようなwarning が出て、音声認識をすることができません。
```
process[static_transformer-1]: started with pid [8267]
process[respeaker_node-2]: started with pid [8268]
process[sound_play-3]: started with pid [8279]
process[speech_to_text-4]: started with pid [8288]
[INFO] [1684061498.679110]: Initializing Respeaker device
[sound_play-3] process has finished cleanly
log file: /home/miyamichi/.ros/log/70bf38a0-e1a3-11ed-a673-a8a1590e9101/sound_play-3*.log
[ERROR] [1684061503.873834]: action 'sound_play' is not initialized.
[INFO] [1684061503.976570]: Respeaker device initialized (Version: 16)
[WARN] [1684061504.512818]: Failed to find respeaker device by name. Using default input
[WARN] [1684061504.514411]: 32 channel is found for respeaker
[WARN] [1684061504.515976]: You may have to update firmware.
```
dmsgでspot体内のPCでデバイスを認識できていることは確認したのですが、nodeで認識できていない感じです。 respeaker_node.pyより↓
```
 # find device                                                                                                                                                                                       
        count = self.pyaudio.get_device_count()
        rospy.logdebug("%d audio devices found" % count)
        for i in range(count):
            info = self.pyaudio.get_device_info_by_index(i)
            name = info["name"].encode("utf-8")
            chan = info["maxInputChannels"]
            rospy.logdebug(" - %d: %s" % (i, name))
            if name.lower().find("respeaker") >= 0:
                self.channels = chan
                self.device_index = i
                rospy.loginfo("Found %d: %s (channels: %d)" % (i, name, chan))
                break
        if self.device_index is None:
            rospy.logwarn("Failed to find respeaker device by name. Using default input")
            info = self.pyaudio.get_default_input_device_info()
            self.channels = info["maxInputChannels"]
            self.device_index = info["index"]
```
cc @mqcmd196

mqcmd196, 2023-05-14:
月曜日に見てみることにします

k-okada, 2023-05-14:
デバイスのパーミッション？
dmsgの全てに情報とls -laかな

nakane11, 2023-05-15:
昨日一緒に見ていて、最初はUSBのパーミッションエラーが出ていたのですが
```
sudo cp -f $(rospack find respeaker_ros)/config/60-respeaker.rules /etc/udev/rules.d/60-respeaker.rules
sudo systemctl restart udev
```
を実行してそのエラーは消えたことを確認しました
dmesgでrespeakerを認識したときのidProductとidVendorは合っていました

mqcmd196, 2023-05-15:
ありがとう，午後に確認します

k-okada, 2023-05-15:
なるほど．
メッセージ見ると
jsk_3rdparty/respeaker_ros at master · jsk-ros-pkg/jsk_3rdparty · GitHub
の3. Update firmware をやるような感じだけど，これは躊躇しているかんじなのかな．
他のロボット実際に認識していて実績のあるデバイスにケーブルつないで試してみると，ハードとソフトの切り分けが出来そうなのと，

            rospy.logdebug(" - %d: %s" % (i, name))

のあたりの表示がどうなっているかだけど，最悪はソースでlogerr としたら目立って表示されるのと，いつもよくわからないんだけど，
http://wiki.ros.org/rosconsole#Configuration
みて rosconsole.config ファイルを書いてみて表示させてみるのが次のデバッグポイントでしょうか．
雰囲気的には

log4j.logger.ros.respeaker_ros=DEBUG

だけど，これではだめな時があるのがむつかしい．．
http://wiki.ros.org/rqt_console
で，このloggerの名前を調べることが多いんだけど，このツールも使い方がむつかしくて，右上の歯車マークをおしえて設定する最初のスタートです．

--
◉ Kei Okada

nakane11, 2023-05-15:
3. Update firmwareは/tmp以下にcloneしたusb_4_mic_arrayでdfu.pyを実行してエラーは出ませんでした

heissereal, 2023-05-15:
spotのaudioのグループに私が入っていないことによるエラーで、グループに入ったら解決しました。大日方さんに直していただきました。
大日方さん、中根さん、岡田先生、ありがとうございました。

closed 2023-05-15 by mqcmd196