jsk-ros-pkg/jsk_3rdparty #508: [melodic] rosからvoicevoxを使えない

sawada10, 2024-07-22:
@k-okada 先生　@iory 先生　@mqcmd196 さん

- https://github.com/iory/jsk_3rdparty/tree/voicevox

矢野倉先生の[こちらのコード](https://github.com/iory/jsk_3rdparty/tree/voicevox)でvoicevoxをROSから使おうとしていますが、catkin buildに失敗します。

実行環境は、Ubuntu18.04, melodicです。
なお、別のUbutnu20.04, noeticではbuildできることは確認できています。

実行手順と結果は以下の通りです。
この場合どのようにすれば良いでしょうか。
```
$ cd ~/seisakuten_ws/src/jsk_3rdparty
$ git remote add yanokura git@github.com:iory/jsk_3rdparty.git
$ git fetch yanokura
$ git checkout remotes/yanokura/voicevox
$ cd ~/seisakuten_ws
$ catkin build voicevox
```

```
$ catkin build voicevox
-------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/melodic
Workspace:                   /home/leus/seisakuten_ws
-------------------------------------------------------------
Build Space:        [exists] /home/leus/seisakuten_ws/build
Devel Space:        [exists] /home/leus/seisakuten_ws/devel
Install Space:      [unused] /home/leus/seisakuten_ws/install
Log Space:          [exists] /home/leus/seisakuten_ws/logs
Source Space:       [exists] /home/leus/seisakuten_ws/src
DESTDIR:            [unused] None
-------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------

-------------------------------------------------------------
WARNING: Your current environment's CMAKE_PREFIX_PATH is
different from the cached CMAKE_PREFIX_PATH used the last
time this workspace was built.

If you want to use a different CMAKE_PREFIX_PATH you should
call `catkin clean` to remove all references to the previous
CMAKE_PREFIX_PATH.

Cached CMAKE_PREFIX_PATH:
	/opt/ros/melodic
Current CMAKE_PREFIX_PATH:
	/home/leus/kashiwagi_ws/devel:/home/leus/toshima_ws/devel:/home/leus/m5stack_ws/devel:/home/leus/nao_ws/devel:/opt/ros/melodic
-------------------------------------------------------------

[build] Found '40' packages in 0.0 seconds.                                                                                                                                                                
[build] Package table is up to date.                                                                                                                                                                       
Starting  >>> voicevox                                                                                                                                                                                     
___________________________________________________________________________________________________________________________________________________________________________________________________________
Errors     << voicevox:make /home/leus/seisakuten_ws/logs/voicevox/build.make.008.log                                                                                                                      
make[3]: 警告: jobserver が利用不可: 今回は -j1 を使います. 親 make ルールに `+' を追加しましょう.
make[3]: 警告: jobserver が利用不可: 今回は -j1 を使います. 親 make ルールに `+' を追加しましょう.
make[3]: 警告: jobserver が利用不可: 今回は -j1 を使います. 親 make ルールに `+' を追加しましょう.
[... 35 lines cut ...]
pip._internal.exceptions.DistributionNotFound: No matching distribution found for dataclasses==0.8 (from -r requirements_melodic.txt (line 10))
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/catkin_virtualenv/venv_lock", line 47, in <module>
    venv.lock(
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin_virtualenv/venv.py", line 143, in lock
    run_command(command, check=True)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin_virtualenv/__init__.py", line 44, in run_command
    return subprocess.run(cmd, *args, **kwargs)
  File "/home/leus/anaconda3/envs/py38/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/leus/seisakuten_ws/build/voicevox/venv/bin/pip-compile', '--no-header', 'requirements_melodic.txt', '--pip-args', '-qq --retries 10 --timeout 30', '-o', '/home/leus/seisakuten_ws/src/jsk_3rdparty/3rdparty/voicevox/requirements.txt']' returned non-zero exit status 1.
make[2]: *** [/home/leus/seisakuten_ws/src/jsk_3rdparty/3rdparty/voicevox/requirements.txt] Error 1
make[1]: *** [CMakeFiles/voicevox_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
cd /home/leus/seisakuten_ws/build/voicevox; catkin build --get-env voicevox | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...........................................................................................................................................................................................................
Failed     << voicevox:make           [ Exited with code 2 ]                                                                                                                                               
Failed    <<< voicevox                [ 18.9 seconds ]                                                                                                                                                     
[build] Summary: 0 of 1 packages succeeded.                                                                                                                                                                
[build]   Ignored:   39 packages were skipped or are blacklisted.                                                                                                                                          
[build]   Warnings:  None.                                                                                                                                                                                 
[build]   Abandoned: None.                                                                                                                                                                                 
[build]   Failed:    1 packages failed.                                                                                                                                                                    
[build] Runtime: 19.0 seconds total.                                                                                                                                                                       
```

sawada10, 2024-07-23:
@mqcmd196 さんに直接見ていただき、buildすることができました。
melodicであることが原因ではなく、python3.8の環境でbuildしようとしていたことが原因でした。
`catkin clean`をして、python2.7の環境で`catkin build voicevox`をし直したところ、buildできるようになりました。

また、buildができない直接の原因ではありませんでしたが、bashrcで複数のworkspaceをsourceしており、warningがでていました。
bashrcに書くのをやめて、今後は毎回必要なworkspaceを自分でsourceするようにしようと思います。

closed 2024-07-23 by sawada10

mqcmd196, 2024-07-23:
> python3.8の環境でbuildしようとしていたことが原因でした。
> catkin cleanをして、python2.7の環境でcatkin build voicevoxをし直したところ、buildできるようになりました。

もとのログを見ると，python2.7のpythonは呼ばれてはいるんだけど，subprocess内のpython3がpython3.8というのがおかしい，という話ではないかな？ここがubuntu 18.04標準のpython3.6である必要がある，というのが僕の考えです．

https://github.com/iory/jsk_3rdparty/blob/3b58ae2351bac93507c57656b114aba080888f0d/3rdparty/voicevox/CMakeLists.txt#L17

sawada10, 2024-07-23:
@mqcmd196 さん、ありがとうございます。
[この行](https://github.com/iory/jsk_3rdparty/blob/3b58ae2351bac93507c57656b114aba080888f0d/3rdparty/voicevox/CMakeLists.txt#L17)でpython3が呼ばれていて、python3.6ではなくpython3.8になっていたのが問題ということでしょうか。

もとのログの最後の方の以下の部分を読むと最後にanaconda3のpython3.8でsubprocess.pyが実行されているような気がします。
condaをこのターミナルではactivateしていないのですが、どうしてanacondaのpython3.8が呼ばれているのでしょうか。
```
File "/home/leus/anaconda3/envs/py38/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['/home/leus/seisakuten_ws/build/voicevox/venv/bin/pip-compile', '--no-header', 'requirements_melodic.txt', '--pip-args', '-qq --retries 10 --timeout 30', '-o', '/home/leus/seisakuten_ws/src/jsk_3rdparty/3rdparty/voicevox/requirements.txt']' returned non-zero exit status 1.
make[2]: *** [/home/leus/seisakuten_ws/src/jsk_3rdparty/3rdparty/voicevox/requirements.txt] Error 1
make[1]: *** [CMakeFiles/voicevox_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
```

mqcmd196, 2024-07-23:
そうです

> condaをこのターミナルではactivateしていないのですが、どうしてanacondaのpython3.8が呼ばれているのでしょうか。
過去にoverlayしているworkspaceをビルドしたときにactivateしていて，そのキャッシュ情報が何らかの形で残っていた，とかでしょうか．いずれにせよさっき僕がcatkin
cleanしてしまっていたのでもう再現不可かもしれませんが...

sawada10, 2024-07-23:
なるほど、記憶が曖昧で申し訳ないのですが、
以前にこのworkspace（seisakuten_ws）でbuildしたときに、condaをactivateしていたかもしれないです。

一度、変な環境でbuildするとキャッシュ情報が残ってしまって、buildに失敗する可能性があって、今回はcatkin cleanでそのキャッシュ情報を消したからbuildできたということですよね。
意図せぬ環境でbuildしないように気をつけます。