jsk-ros-pkg/jsk_3rdparty #519: `ros_speech_recognition` is not released on Noetic/arm64

mqcmd196, 2025-01-09:
I found that I cannot install `ros-noetic-ros-speech-recognition` in the arm64 machine. Could you release it?

It seems that `ros_speech_recognition` can be built in the arm64 machine.

```
obinata@core-io:/tmp/jsk_3rdparty/ros_speech_recognition/build $ cmake ..
CMake Warning (dev) at CMakeLists.txt:2 (project):
  Policy CMP0048 is not set: project() command manages VERSION variables.
  Run "cmake --help-policy CMP0048" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The following variable(s) would be set to empty:

    CMAKE_PROJECT_VERSION
    CMAKE_PROJECT_VERSION_MAJOR
    CMAKE_PROJECT_VERSION_MINOR
    CMAKE_PROJECT_VERSION_PATCH
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Using CATKIN_DEVEL_PREFIX: /tmp/jsk_3rdparty/ros_speech_recognition/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/jsk_coreio_ws/devel;/opt/ros/noetic
-- This workspace overlays: /opt/jsk_coreio_ws/devel;/opt/ros/noetic
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3")
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/lib/python3/dist-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /tmp/jsk_3rdparty/ros_speech_recognition/build/test_results
-- Forcing gtest/gmock from source, though one was otherwise available.
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.10")
-- Using Python nosetests: /usr/bin/nosetests3
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
-- gcc dumpmachine returns aarch64-linux-gnu
-- Using system site packages
-- Using virtualenv to run Python nosetests:
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/jsk_3rdparty/ros_speech_recognition/build
-bash: __git_ps1: command not found
obinata@core-io:/tmp/jsk_3rdparty/ros_speech_recognition/build $ m -j
Scanning dependencies of target ros_speech_recognition_gencfg
Scanning dependencies of target ros_speech_recognition_generate_virtualenv
Scanning dependencies of target ros_speech_recognition_install_trained_data
[ 20%] Generating dynamic reconfigure files from cfg/SpeechRecognition.cfg: /tmp/jsk_3rdparty/ros_speech_recognition/build/devel/include/ros_speech_recognition/SpeechRecognitionConfig.h /tmp/jsk_3rdparty/ros_speech_recognition/build/devel/lib/python3/dist-packages/ros_speech_recognition/cfg/SpeechRecognitionConfig.py
[ 40%] Generate virtualenv in /tmp/jsk_3rdparty/ros_speech_recognition/build/venv
Generating reconfiguration files for SpeechRecognition in ros_speech_recognition
Wrote header file in /tmp/jsk_3rdparty/ros_speech_recognition/build/devel/include/ros_speech_recognition/SpeechRecognitionConfig.h
[ 40%] Built target ros_speech_recognition_gencfg
True
ros_speech_recognition is not found in ['jsk_rqt_plugins', 'collada_urdf', 'rosemacs', 'librviz_tutorial', 'jsk_hark_msgs', 'rqt_runtime_monitor', 'xmlrpcpp', 'class_loader', 'jsk_gui_msgs', 'rqt_launch', 'camera_calibration', 'resource_retriever', 'rosgraph_msgs', 'rosbuild', 'pcl_msgs', 'turtle_tf', 'visualization_msgs', 'pcl_conversions', 'rqt_gui_py', 'smclib', 'pr2_description', 'rqt_rviz', 'rqt_robot_monitor', 'theora_image_transport', 'rosboost_cfg', 'xacro', 'assimp_devel', 'image_view2', 'audio_common_msgs', 'diagnostic_aggregator', 'jskeus', 'checkerboard_detector', 'rqt_py_common', 'fcl', 'controller_manager', 'tf2_ros', 'realtime_tools', 'imagesift', 'rqt_tf_tree', 'rosbag_migration_rule', 'robot_state_publisher', 'rqt_bag', 'rostest', 'gencpp', 'map_server', 'roslang', 'rqt_action', 'urg_node', 'qt_dotgraph', 'rostime', 'octomap_server', 'move_base_msgs', 'roseus', 'laser_assembler', 'genlisp', 'pr2eus', 'self_test', 'nodelet_tutorial_math', 'stereo_image_proc', 'controller_interface', 'diff_drive_controller', 'view_controller_msgs', 'turtle_tf2', 'roscreate', 'urdf_parser_plugin', 'turtle_actionlib', 'rosconsole', 'transmission_interface', 'jsk_perception', 'gazebo_dev', 'diagnostic_common_diagnostics', 'rqt_reconfigure', 'python_qt_binding', 'jsk_footstep_msgs', 'trajectory_msgs', 'angles', 'rqt_robot_dashboard', 'rqt_py_console', 'jsk_tools', 'octomap_ros', 'roscpp_serialization', 'camera_info_manager', 'genpy', 'tf2_kdl', 'compressed_depth_image_transport', 'polled_camera', 'roslz4', 'ruckig', 'rqt_nav_view', 'stage_ros', 'image_rotate', 'people_msgs', 'media_export', 'interactive_markers', 'tf_conversions', 'jsk_tilt_laser', 'filters', 'openni_launch', 'dynamic_reconfigure', 'smach', 'eus_assimp', 'jsk_pcl_ros_utils', 'position_controllers', 'geometry_msgs', 'turtlesim', 'euscollada', 'shape_msgs', 'joint_state_publisher', 'roscpp_tutorials', 'jsk_interactive_marker', 'image_transport', 'topic_tools', 'moveit_ros_occupancy_map_monitor', 'forward_command_controller', 'libuvc_camera', 'laser_filters', 'laser_proc', 'joint_state_controller', 'pluginlib_tutorials', 'qt_gui', 'jsk_recognition_utils', 'jsk_data', 'interactive_marker_tutorials', 'rosnode', 'openni2_launch', 'kdl_conversions', 'rqt_plot', 'geometric_shapes', 'fetch_description', 'control_toolbox', 'rosmake', 'rqt_msg', 'actionlib', 'rosbash', 'libsiftfast', 'webkit_dependency', 'rosservice', 'rosgraph', 'rviz', 'srdfdom', 'gazebo_ros_control', 'smach_msgs', 'catkin', 'rqt_image_view', 'rviz_plugin_tutorials', 'diagnostic_msgs', 'urg_c', 'joint_limits_interface', 'rqt_topic', 'visualization_marker_tutorials', 'jsk_topic_tools', 'rviz_python_tutorial', 'object_recognition_msgs', 'pr2_msgs', 'rqt_dep', 'rqt_gui', 'urdf_sim_tutorial', 'euslisp', 'bondpy', 'nav_msgs', 'qwt_dependency', 'moveit_ros_planning', 'image_publisher', 'tf2_geometry_msgs', 'image_view', 'robot_self_filter', 'rosclean', 'gazebo_plugins', 'openni_camera', 'rosout', 'image_geometry', 'map_msgs', 'rosmsg', 'nodelet_topic_tools', 'message_runtime', 'joint_state_publisher_gui', 'actionlib_msgs', 'cv_bridge', 'bond', 'audio_video_recorder', 'resized_image_transport', 'diagnostic_analysis', 'rqt_bag_plugins', 'std_msgs', 'jsk_recognition_msgs', 'ros_environment', 'camera_calibration_parsers', 'mk', 'rqt_robot_steering', 'posedetection_msgs', 'roslaunch', 'multi_map_server', 'sensor_msgs', 'urdf', 'rqt_gui_cpp', 'actionlib_tutorials', 'tf2_msgs', 'rqt_service_caller', 'roslint', 'sound_play', 'cpp_common', 'tf2_py', 'depth_image_proc', 'pybind11_catkin', 'roslib', 'jsk_pcl_ros', 'pr2_controllers_msgs', 'rqt_web', 'kdl_parser', 'rqt_pose_view', 'rqt_console', 'control_msgs', 'tf2_eigen', 'roscpp', 'urdfdom_py', 'jsk_rviz_plugins', 'moveit_msgs', 'rosconsole_bridge', 'moveit_ros_perception', 'rosbag', 'rqt_graph', 'rosmaster', 'rosbag_storage', 'jsk_network_tools', 'rostopic', 'tf', 'gazebo_msgs', 'qt_gui_py_common', 'opencv_apps', 'stereo_msgs', 'tf2', 'pcl_ros', 'cmake_modules', 'compressed_image_transport', 'rqt_top', 'random_numbers', 'rospack', 'dynamic_tf_publisher', 'roswtf', 'gl_dependency', 'pr2_mechanism_msgs', 'moveit_core', 'octomap_msgs', 'genmsg', 'diagnostic_updater', 'rospy', 'smach_ros', 'message_generation', 'std_srvs', 'rgbd_launch', 'virtual_force_publisher', 'bondcpp', 'image_proc', 'nodelet', 'gennodejs', 'eigen_stl_containers', 'stage', 'rqt_srv', 'eigen_conversions', 'octomap', 'qt_gui_cpp', 'rqt_publisher', 'hardware_interface', 'urdf_tutorial', 'roslisp', 'message_filters', 'rqt_shell', 'roscpp_traits', 'pluginlib', 'rospy_tutorials', 'controller_manager_msgs', 'rqt_moveit', 'openni2_camera', 'rqt_logger_level', 'laser_geometry', 'gazebo_ros', 'rosunit', 'collada_parser', 'rosparam', 'geneus', 'catkin_virtualenv', 'vision_msgs', 'global_planner', 'rotate_recovery', 'rosauth', 'open_karto', 'julius_ros', 'pr2_common_action_msgs', 'pr2_mechanism_diagnostics', 'fetch_navigation', 'mbf_costmap_core', 'eusurdf', 'joint_trajectory_generator', 'fetch_auto_dock_msgs', 'fetch_maps', 'fingertip_pressure', 'robot_localization', 'pr2_machine', 'geographic_msgs', 'naoqi_libqicore', 'audio_play', 'moveit_python', 'naoqi_libqi', 'clear_costmap_recovery', 'simple_grasping', 'moveit_ros_warehouse', 'julius', 'rwt_image_view', 'pr2_computer_monitor', 'base_local_planner', 'amcl', 'pr2eus_moveit', 'ddynamic_reconfigure', 'twist_mux', 'moveit_ros_planning_interface', 'move_slow_and_clear', 'slam_karto', 'pr2_dashboard_aggregator', 'costmap_2d', 'switchbot_ros', 'velodyne_msgs [... cut ...] sk_robot_startup']
Package [ros_speech_recognition] is not found in current workspace. Skipping download
[ 40%] Built target ros_speech_recognition_install_trained_data
[ 60%] Install requirements to /tmp/jsk_3rdparty/ros_speech_recognition/build/venv
ERROR: launchpadlib 1.10.13 requires testresources, which is not installed.
[ 80%] Prepare relocated virtualenvs for develspace and installspace
[100%] Per-package virtualenv target
[100%] Built target ros_speech_recognition_generate_virtualenv
```

k-okada, 2025-01-09:
this one is currently disabled -> https://build.ros.org/job/Nbin_ufv8_uFv8__ros_speech_recognition__ubuntu_focal_arm64__binary/

Please check if this works, not `catkin build` but `dpkg-buildpakcage -b -us -uc` using folloiwng source tree,
https://github.com/tork-a/jsk_3rdparty-release/blob/debian/noetic/focal/ros_speech_recognition/

and ask to remove from balck list.
https://github.com/ros-infrastructure/ros_buildfarm_config/blob/c8f349e3926cd856586731d09b21c0bb7321de4e/noetic/release-arm64-build.yaml#L27

mqcmd196, 2025-01-09:
I see, I'll check it debian build env later

k-okada, 2025-01-09:
if that fails, please try https://github.com/v4hn/ros-deb-builder-action/pull/3/files#diff-4d2a8eefdf2a9783512a35da4dc7676a66404b6f3826a8af9aad038722da6823R156-R158

mqcmd196, 2025-01-10:
Thank you. I forget

> https://github.com/jsk-ros-pkg/jsk_3rdparty/issues/494#issuecomment-1674274910 (confirmed by @mqcmd196 )

comment and maybe I didn't know how to fix it at that time.

> if that fails, please try https://github.com/v4hn/ros-deb-builder-action/pull/3/files#diff-4d2a8eefdf2a9783512a35da4dc7676a66404b6f3826a8af9aad038722da6823R156-R158

It worked and I added a comment.

mqcmd196, 2025-01-10:
command result

[typescript.txt](https://github.com/user-attachments/files/18369848/typescript.txt)

mqcmd196, 2025-01-10:
I'm not sure but adding the patch like https://github.com/tork-a/darknet-release/tree/master/patches ?