jsk-ros-pkg/jsk_3rdparty #535: voicevoxで使えるキャラクターの声を増やしたい

sawada10, 2025-04-23:
@iory @k-okada 

ROSのvoicevoxで今使える声の種類が、[README](https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/master/3rdparty/voicevox/README.md)にあげてくださっているものかと思うのですが、さらに増やすことは可能でしょうか？

柏木さんの声を「[ちび式じい](https://voicevox.hiroshiba.jp/dormitory/chibishikiji/)」にしたいと思っています。（たぶん@ayaha-nも将来的に使う気がしています）
[voicevoxのgithub](https://github.com/VOICEVOX/voicevox/blob/b24884608e871bf990c3e900f41c87b762efd6ba/public/updateInfos.json#L480)を確認したところ、updateInfos.jsonにちび式じいを追加した履歴があり、そちらでは対応していそう？なのですが、
どこを編集したら良いのかがわからず、教えていただけると嬉しいです。

mqcmd196, 2025-04-23:
特に何もしなくても使えると思います．合成したい声を引数に入れて試してみてください

sawada10, 2025-04-23:
うそでした . 修正中です by obinata

sawada10, 2025-04-23:
@mqcmd196 
https://github.com/sawada10/jsk_3rdparty/tree/add-chibishikijii-voice

先程、@mqcmd196が作業してくださっていたものです

mqcmd196, 2025-05-01:
close via https://github.com/jsk-ros-pkg/jsk_3rdparty/pull/539

closed 2025-05-01 by mqcmd196