jsk-ros-pkg/jsk_3rdparty #541: [Noetic] voicevoxのlaunchを立ち上げるとrequests moduleがないというエラーが出る

sawada10, 2025-05-12:
@k-okada @iory @mqcmd196 

私の環境の問題な気がするのですが、jsk_3rdpartyの最新master（tag: [2.1.30](https://github.com/jsk-ros-pkg/jsk_3rdparty/releases/tag/2.1.30)) でvoicevoxのlaunchを立ち上げると、以下のようなエラーが出ました。

```
$ roslaunch voicevox voicevox_texttospeech.launch 
... logging to /home/leus/.ros/log/186c1580-2f24-11f0-b0f1-89906e5f4c14/roslaunch-leus-ThinkPad-T480s-22604.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leus-ThinkPad-T480s:46059/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    sound_play_jp (sound_play/soundplay_node.py)
    voicevox_server (voicevox/run-voicevox)

auto-starting new master
process[master]: started with pid [22619]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 186c1580-2f24-11f0-b0f1-89906e5f4c14
process[rosout-1]: started with pid [22636]
started core service [/rosout]
process[voicevox_server-2]: started with pid [22643]
process[sound_play_jp-3]: started with pid [22646]
Traceback (most recent call last):
  File "/home/leus/kashiwagi_ws/devel/.private/voicevox/share/voicevox/voicevox_engine/run.py", line 19, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
[voicevox_server-2] process has died [pid 22643, exit code 1, cmd /home/leus/kashiwagi_ws/devel/share/voicevox/bin/run-voicevox --voicelib_dir=/home/leus/kashiwagi_ws/devel/share/voicevox/voicevox_core --host localhost --port 50021 --cpu_num_threads=1 --load_all_models -- __name:=voicevox_server __log:=/home/leus/.ros/log/186c1580-2f24-11f0-b0f1-89906e5f4c14/voicevox_server-2.log].
log file: /home/leus/.ros/log/186c1580-2f24-11f0-b0f1-89906e5f4c14/voicevox_server-2*.log
[voicevox_server-2] restarting process
process[voicevox_server-2]: started with pid [22666]
/opt/ros/noetic/lib/sound_play/soundplay_node.py:330: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()
/opt/ros/noetic/lib/sound_play/soundplay_node.py:331: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.g_loop = threading.Thread(target=GObject.MainLoop().run)
Traceback (most recent call last):
  File "/home/leus/kashiwagi_ws/devel/.private/voicevox/share/voicevox/voicevox_engine/run.py", line 19, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
[voicevox_server-2] process has died [pid 22666, exit code 1, cmd /home/leus/kashiwagi_ws/devel/share/voicevox/bin/run-voicevox --voicelib_dir=/home/leus/kashiwagi_ws/devel/share/voicevox/voicevox_core --host localhost --port 50021 --cpu_num_threads=1 --load_all_models -- __name:=voicevox_server __log:=/home/leus/.ros/log/186c1580-2f24-11f0-b0f1-89906e5f4c14/voicevox_server-2.log].
log file: /home/leus/.ros/log/186c1580-2f24-11f0-b0f1-89906e5f4c14/voicevox_server-2*.log
[voicevox_server-2] restarting process
process[voicevox_server-2]: started with pid [22684]
Traceback (most recent call last):
  File "/home/leus/kashiwagi_ws/devel/.private/voicevox/share/voicevox/voicevox_engine/run.py", line 19, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
[voicevox_server-2] process has died [pid 22684, exit code 1, cmd /home/leus/kashiwagi_ws/devel/share/voicevox/bin/run-voicevox --voicelib_dir=/home/leus/kashiwagi_ws/devel/share/voicevox/voicevox_core --host localhost --port 50021 --cpu_num_threads=1 --load_all_models -- __name:=voicevox_server __log:=/home/leus/.ros/log/186c1580-2f24-11f0-b0f1-89906e5f4c14/voicevox_server-2.log].
log file: /home/leus/.ros/log/186c1580-2f24-11f0-b0f1-89906e5f4c14/voicevox_server-2*.log
[voicevox_server-2] restarting process
process[voicevox_server-2]: started with pid [22696]
[INFO] [1747049255.114195]: Loading from plugin definitions
Traceback (most recent call last):
  File "/home/leus/kashiwagi_ws/devel/.private/voicevox/share/voicevox/voicevox_engine/run.py", line 19, in <module>
    import requests
ModuleNotFoundError: No module named 'requests'
```

確認したことは以下のとおりです
1. `catkin clean voicevox`をしてから、`rosdep install --ignore-src --from-paths . -y -r`や`catkin build`をしたが、同じエラーがでる
2. `requests`モジュールの存在を確認
```
$ pip list |grep requests
requests                      2.22.0
requests-oauthlib             1.0.0
requests-unixsocket           0.2.0
```
`requests`は入っていそう。試しに以下のように`import`を試してもエラーは出ず、入っていそう。
```
$ python3
Python 3.8.10 (default, Mar 18 2025, 20:04:55) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import requests
>>>
```

その他確認すべき事項などをご教示いただけると嬉しいです。

iory, 2025-05-12:
以下を試してみてどうなるかを教えてください。

```
roscd voicevox 
git clean -dfx .
cd ~/kashiwagi_ws
catkin clean --all -y
source /opt/ros/noetic/setup.bash
catkin build voicevox
```

sawada10, 2025-05-13:
@iory 
ありがとうございます。
教えていただいた手順を実行したところ、動くようになり、しゃべることも確認できました。
これは、以前buildしたときに生成されたファイル（= .gitignore以下のもの）が残っていて、どこかでそれを参照してしまっていたため、エラーが出ていたということでしょうか？

```
$ roslaunch voicevox voicevox_texttospeech.launch 
... logging to /home/leus/.ros/log/11b5556a-2fbc-11f0-b0f1-89906e5f4c14/roslaunch-leus-ThinkPad-T480s-27287.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://leus-ThinkPad-T480s:40513/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    sound_play_jp (sound_play/soundplay_node.py)
    voicevox_server (voicevox/run-voicevox)

auto-starting new master
process[master]: started with pid [27302]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 11b5556a-2fbc-11f0-b0f1-89906e5f4c14
process[rosout-1]: started with pid [27319]
started core service [/rosout]
process[voicevox_server-2]: started with pid [27326]
process[sound_play_jp-3]: started with pid [27327]
/opt/ros/noetic/lib/sound_play/soundplay_node.py:330: PyGIDeprecationWarning: Since version 3.11, calling threads_init is no longer needed. See: https://wiki.gnome.org/PyGObject/Threading
  GObject.threads_init()
/opt/ros/noetic/lib/sound_play/soundplay_node.py:331: PyGIDeprecationWarning: GObject.MainLoop is deprecated; use GLib.MainLoop instead
  self.g_loop = threading.Thread(target=GObject.MainLoop().run)
[INFO] [1747114527.062622]: Loading from plugin definitions
[INFO] [1747114527.568685]: sound_play node is ready to play sound
INFO:     Started server process [27326]
INFO:     Waiting for application startup.
Downloading: "https://github.com/r9y9/open_jtalk/releases/download/v1.11.1/open_jtalk_dic_utf_8-1.11.tar.gz"
dic.tar.gz: 100%|███████████████████████████████████████████████| 22.6M/22.6M [00:02<00:00, 8.72MB/s]
Extracting tar file /home/leus/kashiwagi_ws/devel/share/voicevox/venv/lib/python3.8/site-packages/pyopenjtalk/dic.tar.gz
reading /home/leus/.local/share/voicevox-engine-dev/tmpapdcq21o ... 73
emitting double-array: 100% |###########################################| 

done!
INFO:     Application startup complete.
INFO:     Uvicorn running on http://localhost:50021 (Press CTRL+C to quit)
```

iory, 2025-05-13:
良かったです

catkin virtual envを使うとrequirements.txtというものがvoicevox以下に作成されるので前回build時に途中で停止したとか、git logの古いものが残っていてそれでうまくいかなかったのかと推察してます。

sawada10, 2025-05-13:
なるほど、次回同様のことが起きた時に参考にして対処してみようと思います。
ありがとうございます！

closed 2025-05-13 by sawada10